NVIDIAGameWorks/PhysX #344: Joint position encoding for articulations

cfognom, 2020-10-04:
If I have a 3 dof reduced coordinate articulation joint, what is the encoding of the joint positions? I'm using PhysX though Unity and have tested all 6 possible Tait–Bryan rotation orderings to reproduce the joint positions of the articulation, but neither of them were accurate. I also found an earlier post about this issue, saying that the ordering was (ZYX format, Tait-bryan) but according to my tests it is not. Why is this not specified in the docs?

lasagnaphil, 2020-10-29:
The spherical joint format abruptly changed in an update in 4.1.1. The joint rotations are now embedded in a rotation vector (axis-angle representation). I guess this change was necessary to avoid singularities in the previous Euler angle representation, but it was quite frustrating that this wasn't anywhere in the docs or the changelogs.

cfognom, 2020-10-29:
Ah, yes axis angle accurately reproduces their rotations. Thank you!

closed 2020-10-29 by cfognom