NVIDIAGameWorks/PhysX #349: [Physx4.1] raycast results are not filtered by maxDist parameter

Esentiel, 2020-10-13:
Hello!

The problem is, I pass some small value for maxDist (1-5) and expect that there will no be resulted hits with hit.distance more than the value I passed to the PxGeometryQuery::raycast() function.
I debugged the PhysX a bit and didn't find out the code that should shrink results based on the distance to a hit. The distance was used in filtration before even hit detection in "SLABS_TEST2" macro and that's pretty much all I was able to find.
Here is callstack:
```
PhysXCommon_64.dll!RayTriOverlapT<RayParams>(physx::PxRaycastHit & mStabbedFace, const physx::PxVec3 & vert0, const physx::PxVec3 & vert1, const physx::PxVec3 & vert2, const RayParams * params) Line 134	C++
PhysXCommon_64.dll!`anonymous namespace'::LeafFunction_RaycastCB::doLeafTest(`anonymous-namespace'::RayParamsCB * params, unsigned int primIndex) Line 512	C++
PhysXCommon_64.dll!BV4_ProcessStreamKajiyaNoOrderQ<0,`anonymous namespace'::LeafFunction_RaycastCB,A0x62e1568b::RayParamsCB>(const physx::Gu::BVDataPackedT<physx::Gu::QuantizedAABB> * node, unsigned int initData, `anonymous-namespace'::RayParamsCB * params) Line 226	C++
PhysXCommon_64.dll!processStreamRayNoOrder<0,`anonymous namespace'::LeafFunction_RaycastCB,A0x62e1568b::RayParamsCB>(const physx::Gu::BV4Tree & tree, `anonymous-namespace'::RayParamsCB * params) Line 84	C++
PhysXCommon_64.dll!BV4_RaycastCB(const physx::PxVec3 & origin, const physx::PxVec3 & dir, const physx::Gu::BV4Tree & tree, const physx::PxMat44 * worldm_Aligned, float maxDist, float geomEpsilon, unsigned int flags, physx::Gu::HitCode(*)(void *, const physx::PxVec3 &, const physx::PxVec3 &, const physx::PxVec3 &, unsigned int, float, float, float) callback, void * userData) Line 544	C++
PhysXCommon_64.dll!raycastVsMeshCB(const physx::Gu::BV4Tree & tree, const physx::PxVec3 & orig, const physx::PxVec3 & dir, float maxDist, float geomEpsilon, bool doubleSided, physx::Gu::HitCode(*)(void *, const physx::PxVec3 &, const physx::PxVec3 &, const physx::PxVec3 &, unsigned int, float, float, float) callback, void * userData) Line 269	C++
PhysXCommon_64.dll!physx::Gu::raycast_triangleMesh_BV4(const physx::Gu::TriangleMesh * mesh, const physx::PxTriangleMeshGeometry & meshGeom, const physx::PxTransform & pose, const physx::PxVec3 & rayOrigin, const physx::PxVec3 & rayDir, float maxDist, physx::PxFlags<enum physx::PxHitFlag::Enum,unsigned short> hitFlags, unsigned int maxHits, physx::PxRaycastHit * hits) Line 482	C++
PhysXCommon_64.dll!physx::Gu::Midphase::raycastTriangleMesh(const physx::Gu::TriangleMesh * mesh, const physx::PxTriangleMeshGeometry & meshGeom, const physx::PxTransform & meshTransform, const physx::PxVec3 & rayOrigin, const physx::PxVec3 & rayDir, float maxDist, physx::PxFlags<enum physx::PxHitFlag::Enum,unsigned short> hitFlags, unsigned int maxHits, physx::PxRaycastHit * hits) Line 318	C++
PhysXCommon_64.dll!raycast_triangleMesh(const physx::PxGeometry & geom, const physx::PxTransform & pose, const physx::PxVec3 & rayOrigin, const physx::PxVec3 & rayDir, float maxDist, physx::PxFlags<enum physx::PxHitFlag::Enum,unsigned short> hitFlags, unsigned int maxHits, physx::PxRaycastHit * hits) Line 378	C++
PhysXCommon_64.dll!physx::PxGeometryQuery::raycast(const physx::PxVec3 & rayOrigin, const physx::PxVec3 & rayDir, const physx::PxGeometry & geom, const physx::PxTransform & pose, float maxDist, physx::PxFlags<enum physx::PxHitFlag::Enum,unsigned short> hitFlags, unsigned int maxHits, physx::PxRaycastHit * rayHits) Line 215	C++
```
At this point hit with distance more than I did set is added to the results.

Can you please help me a bit to find out what is the reason for such behavior or please point to the code that should do the job, cut the hits by max distance parameter.

Here are parameters wich were used to invoke PxGeometryQuery::raycast() function, just in case:
```
+		rayOrigin	{x=-24.6143074 y=1.24935055 z=0.967061043 }	const physx::PxVec3 &
+		rayDir	{x=-0.578418732 y=-0.0491725728 z=-0.814256608 }	const physx::PxVec3 &
+		geom	{mType=eTRIANGLEMESH (5) }	const physx::PxGeometry &
+		pose	{q={x=0.000000000 y=0.000000000 z=0.000000000 ...} p={x=0.000000000 y=0.000000000 z=0.000000000 } }	const physx::PxTransform &
		maxDist	1.00000000	float
+		hitFlags	{mBits=139 }	physx::PxFlags<enum physx::PxHitFlag::Enum,unsigned short>
		maxHits	8	unsigned int
////
        mStabbedFace.distance = 4.87200403 // that's the actual distance to the hit point
```

Thanks in advance!

Esentiel, 2020-10-16:
> Does your mesh have a non-uniform scale?

It has uniform scale. Basically, it's a huge terrain and raycast starts from players weapon (who's standing on this terrain) in the direction to this very terrain. And this bug is reproduced only with raycasting into terrain trimesh.

Pierre-Terdiman, 2020-10-16:
I see that the ray-vs-mesh function has several codepaths, and the one taken for scaled mesh has this suspicious note:

https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/source/geomutils/src/mesh/GuBV4_Raycast.cpp#L518

So there might be a bug here, a missing bit that I forgot about. I think you have two workarounds meanwhile:

a) bake the scale into the mesh (i.e. scale the mesh vertices, then use an identity scale for that mesh). It should go to another codepath which shouldn't have this issue.

b) use the BVH33 format for this mesh, instead of BVH34. It's an entirely different data-structure with an entirely different codepath again, and that one has been shipped for a longer time so it's unlikely to have this issue.

Pierre-Terdiman, 2020-10-16:
Note: it's perfectly legal to use both BVH33 and BVH34 formats in the same scene. You can use BVH33 just for this scaled mesh, for example.

Esentiel, 2020-10-16:
Yeah it seems it has non-uniform scale. I beg your pardon, in one place it was set as uniform scaled (in user code, not in PhysX) but in PhysX it appears to be scaled indeed.
Thanks for the quick response, I will test suggested work-arounds and maybe will try to prepare pull request with possible fix for this. 
Please do not close the ticket yet.

closed 2020-10-22 by Esentiel

Pierre-Terdiman, 2020-10-28:
I think the fix is just to add this missing line in LeafFunction_RaycastCB :

`if(StabbedFace.distance<params->mStabbedFace.mDistance)`

This is consistent with what we do in other places. So the code becomes something like this:

```
class LeafFunction_RaycastCB
{
public:
	static PxIntBool doLeafTest(RayParamsCB* PX_RESTRICT params, PxU32 primIndex)
	{
		PxU32 nbToGo = getNbPrimitives(primIndex);
		do
		{
			PxU32 VRef0, VRef1, VRef2;
			getVertexReferences(VRef0, VRef1, VRef2, primIndex, params->mTris32, params->mTris16);

			const PxVec3& p0 = params->mVerts[VRef0];
			const PxVec3& p1 = params->mVerts[VRef1];
			const PxVec3& p2 = params->mVerts[VRef2];

			PX_ALIGN_PREFIX(16)	char buffer[sizeof(PxGeomRaycastHit)] PX_ALIGN_SUFFIX(16);
			PxGeomRaycastHit& StabbedFace = reinterpret_cast<PxGeomRaycastHit&>(buffer);
			if(RayTriOverlapT<RayParams>(StabbedFace, p0, p1, p2, params))
			{
				**if(StabbedFace.distance<params->mStabbedFace.mDistance)**
				{
					HitCode Code = (params->mCallback)(params->mUserData, p0, p1, p2, primIndex, StabbedFace.distance, StabbedFace.u, StabbedFace.v);
					if(Code==HIT_EXIT)
						return 1;
				}

```

Esentiel, 2020-10-29:
Hi!
For our project we simply implemented filtration by distance in PhysX wrapper. I can create pull request with above changes if are willing to merge it nevertheless.

Pierre-Terdiman, 2020-10-29:
Sure, no problem, your choice. But it's a proper bug that we had to fix anyway so I submitted the above change to our trunk yesterday. Thank you for the report :)