NVIDIAGameWorks/PhysX #350: Unable to drive PxArticulationJointReducedCoordinate

mechanicalfuture, 2020-10-14:
Hi

I'm having some trouble with the RC articulation joint drives and I was hoping you might have some pointers as to what's going wrong.

I've been able to isolate the problem in `physx/snippets/snippetarticulation/SnippetArticulation.cpp` by reducing the scissor lift
to only the base with the two "root" blocks (where the right root is the prismatic joint drive).

After commenting out the scissors and top boxes (L136-L286, L290- L362) it looks like so,

[image]

After making this change the joint drive is still functional. Until. After modifying the `stepPhysics()` function such that the drive target is no longer being ramped at each step and instead is only updated every t > 1 seconds: 
```
#if USE_REDUCED_COORDINATE_ARTICULATION
	static bool gClosing = true;
	static std::chrono::steady_clock::time_point tlast = std::chrono::steady_clock::now();
#endif

void stepPhysics(bool /*interactive*/)
{
	const PxReal dt = 1.0f / 60.f;
#if USE_REDUCED_COORDINATE_ARTICULATION & CREATE_SCISSOR_LIFT
	PxReal driveValue = gDriveJoint->getDriveTarget(PxArticulationAxis::eZ);

	auto tnow = std::chrono::steady_clock::now();
	std::chrono::duration<double> diff = tnow - tlast;

	if (diff.count() > 1.0) {
		tlast = tnow;

		if (gClosing && driveValue < -1.2f)
			gClosing = false;
		else if (!gClosing && driveValue > 0.f)
			gClosing = true;

		if (gClosing)
			driveValue -= dt * 0.25f;
		else
			driveValue += dt * 0.25f;
	}

	gDriveJoint->setDriveTarget(PxArticulationAxis::eZ, driveValue);
#endif

	gScene->simulate(dt);
	gScene->fetchResults(true);
}
```

If this ramping delay is less than 0.1 seconds the joint drives Ok, anything greater than 0.3 seconds and it fails to move (also note that it makes no difference here if `setDriveTarget` is called each step or not).

In most robotic applications, the joint targets are likely to a have a constant target for some duration, so this setup mirrors my application well. 

Any advice with quick turnarounds would be greatly appreciated!!!

preist-nvidia, 2020-10-14:
Sounds to me like it could be an issue with the RC articulation going to sleep. Could you please try setting either a high wake counter via the articulation's setWakeCounter method, or just calling the wakeUp method when you set the new target?

mechanicalfuture, 2020-10-14:
Yes, that was it thank you!!

closed 2020-10-14 by mechanicalfuture