NVIDIAGameWorks/PhysX #351: Unrealistic Bouncing Ball

alanjfs, 2020-10-15:
Hi all,

I've been noticing this odd behaviour with restitution.

[image]

Notice how that last bounce seems to absorb all remaining energy?

Here's what I'd like to have happen.

- https://www.youtube.com/watch?v=aY3TrpiUOqE

Notice how at the end the frequency of bounces increase until it eventually bounces at maybe 100 bounces per second before stopping.

The above simulation was run with 0 dynamic/static friction on both ground and ball, and 1.0 restitution on ground and 0.5 restitution on ball.

Is there some attribute I can set to mimic the YouTube footage?

Thanks.

kstorey-nvidia, 2020-10-15:
Please experiment adjusting the bounce threshold on the scene. This defines a relative velocity, below which restitution is ignored. It is defaulted to 2m/s, but can be reduced to get more realistic results.

You will likely see some simulation stability artefacts if you set bounce threshold smaller than the velocity caused by gravity over a single simulation step.

alanjfs, 2020-10-15:
> Please experiment adjusting the bounce threshold on the scene. This defines a relative velocity, below which restitution is ignored. It is defaulted to 2m/s, but can be reduced to get more realistic results.

Thanks for this, I didn't know about that threshold. Will experiment and report back!

alanjfs, 2020-10-15:
Hm, yes indeed I can see how it helps with stability. But I can't seem to find a value that behaves well.

Here's 100%, 50% and 10% of the default value.

Any other suggestions?
[image]

alanjfs, 2020-10-15:
The above examples were made with TGS, and it looks like PGS yields a better results in this particular instance. They also had a slight linear and angular damping which I removed.

Here's the equivalent PGS example at 10% default value.

[image]

I can work with this. There's still a minor jolt on the final bounce that I wish wasn't there, so if there's any other setting you think might help I would love to hear it. In any case, I'm happy to close this issue, thanks for the help!

kstorey-nvidia, 2020-10-15:
In order to simulate very high frequency bounces, you will probably have to simulate at smaller time-steps. Then you should be able to reduce the bounce threshold further and still get stable results. This will negatively impact performance.

Regarding the behaviour you observed with the TGS solver. The TGS solver may yield better results in this particular case if you set velocity iterations to 0 and just use position iterations. There will be improvements coming in the next major PhysX release that address this kind of artefact and yield the same/better results than the PGS solver. 

Note that trying to set velocity iterations to 0 will issue a warning/refuse to set that value in debug/checked builds in the current PhysX release. The terms "velocity" and "position" iterations are a bit misleading because they are actually both solving for velocity. One solves a constraint that includes error correction terms and the other solves one without error correction terms.

alanjfs, 2020-10-15:
Yes, I should have mentioned. This is solving at a much smaller timestep than the initial examples as well, that makes sense.

>  The TGS solver may yield better results in this particular case if you set velocity iterations to 0 and just use position iterations

Interesting, thanks I will experiment with that.

> There will be improvements coming in the next major PhysX release that address this kind of artefact and yield the same/better results than the PGS solver.

Very good news!

> The terms "velocity" and "position" iterations are a bit misleading because they are actually both solving for velocity. One solves a constraint that includes error correction terms and the other solves one without error correction terms.

Interesting, I wasn't aware of that. Thanks @kstorey-nvidia I've got lots to experiment with now.

alanjfs, 2020-10-15:
> The terms "velocity" and "position" iterations are a bit misleading

What would be a better choice of words for these? E.g. `preErrorCorrectionIterations`? :S

alanjfs, 2020-10-28:
Closing this, I got what I was looking for. Thanks again!

closed 2020-10-28 by alanjfs