NVIDIAGameWorks/PhysX #353: Vehicle example in real time

yumianhuli2, 2020-10-17:
Hello PhysX Team！

Are the vehicle examples in the Snippet folder is real time simulation?

In these examples, timestep is 1/60. What is the conversion relation between this and real time?

How does PhysX guarantee real-time in these examples? Are there any examples of a complete cycle of computation instead of in real time?

Thanks！

gyeomannvidia, 2020-10-21:
The function stepPhysics() is just a wrapper for the PhysX simulation function calls. 

PhysX will try to advance the simulation forwards in time by the user-specified timestep as fast as it can.   PhysX only considers simulation time and does not track computer clock time.

yumianhuli2, 2020-10-22:
> The function stepPhysics() is just a wrapper for the PhysX simulation function calls.
> 
> PhysX will advance the simulation forwards in time by the user-specified timestep as fast as it can. PhysX only considers simulation time and does not track computer clock time.
> 
> If you want a real-time simulation of vehicles then a separate system is required. This system needs to track the passing of real time and update the physx scene accordingly so that simulation time matches real time.

OK，It's like updating with render frames！

closed 2020-10-23 by yumianhuli2