NVIDIAGameWorks/PhysX #355: Possibility of PhysX 5.1 GPU UE4 custom branch?

SephirothFFKH, 2020-10-22:
Since UE4 is transitioning to Chaos Physics which, in my opinion, look much more basic compared to all the new features of PhysX 5.1, would it be possible for Nvidia to have a PhysX 5.1 UE4 custom branch with GPU acceleration. I do now that Nvidia has unfortunately stopped maintaining all their simulation oriented custom branches such as Blast, Flex, Flow and etc. but could you make an exception for a potential UE4 PhysX 5.1 GPU branch?

Thank you

Zeblote, 2020-10-25:
Well that's not very useful while PhysX 5.1 is not released is it

oscarbg, 2020-10-29:
first has to come 5.0 announced at start of the year..

Hoikas, 2020-11-23:
Unfortunately, NVIDIA appears to no longer be interested in maintaining PhysX, as demonstrated by the large list of unanswered pull requests.

Zeblote, 2020-11-26:
Oh they are maintaining it, considering PhysX 5.1 exists and is used in the Omniverse stuff.

What they don't seem to be interested in anymore, is sharing it.... coming early 2020 yeah right...