NVIDIAGameWorks/PhysX #356: Odd behavior with linearLimit and PxD6Joint

alanjfs, 2020-10-28:
Hi all,

I'm looking at an odd result I can't make sense out of.

[image]

I've simplified the setup with a minimal amount of moving parts.

[image]

Which, as you can see, there are a total of 3 rigid bodies and 2 constraints.

1. Two kinematic rigids for the upper and lower arm
2. One dynamic rigid for the "muscle"
3. Two constraints pulling it in either direction, towards the muscle attachment points on both arm rigids

I'll attach a PVD session file here as well, where you can inspect the attributes more closely.

[image]

[madmuscle.zip](https://github.com/NVIDIAGameWorks/PhysX/files/5450813/madmuscle.zip)

What I'm expecting to see, is for these two constraints to pull the muscle in a straight line towards the end points. But instead, it somehow gets mad amounts of angular force applied to it as the lower arm moves. Why? What's happening?

alanjfs, 2020-10-28:
Adding to this, this only happens with the TGS solver, PGS behaves as I would expect. Other parts of my simulation run more stable with TGS, so is there any way of getting TGS to play along here?

[image]

alanjfs, 2020-10-28:
I take that back; with a little more linear stiffness it's back. The problem must be elsewhere, maybe in how the constraints are set up?

[image]

alanjfs, 2020-10-28:
I wanted to have a look at this `eACCELERATION` flag mentioned in [the docs](https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/Joints.html#limits):

> When using spring limits, the eACCELERATION flag is strongly recommended. This flag will automatically scale the strength of the spring according to the masses and inertias of objects that the limit is acting upon, and can substantially reduce the amount of tuning required for good, stable behavior.

But what is it, where can I find it and where do I apply it?

There's one for drives, but the recommendation clearly specifies spring *limits*. But neither the `PxSpring`, `PxJointLinearLimitPair` or `PxD6Joint` seem to take any flag by that name. :S

alanjfs, 2020-10-28:
Some more evidence, where I can get PGS to play ball, but TGS is completely out of it.

- [pgstgs.zip](https://github.com/NVIDIAGameWorks/PhysX/files/5452013/pgstgs.zip)

[image]

Here I also discovered that if I `PxD6Joint.setDistanceLimit()` instead of `.setLinearLimit()` I can get it drawn in PVD.

What's more interesting is that if I lower the stiffness and increase damping with TGS, I can get it to stop jittering; but then instead behave interestingly as one of the constrained kinematic rigids move.

[image]

See how it follows in the direction of the left-hand kinematic rigid, and then slowly returns back to where I expected it to go? Why is that? How do I avoid that? Is it related to the overall problem of stability?

kstorey-nvidia, 2020-10-28:
I took a very brief look at the PVD capture for the original case. In that case, the inertia tensor was not set up on the larger capsule( it had just 1,1,1), which would have hurt stability.

For the other cases, if it involves kinematics, this may be a bugfix that has not yet made it into a public release. I just checked and that fix is not currently in PhysX 4.1.

This line:

if(isKinematic0)
			s.velTarget -= vel0;
		if(isKinematic1)
			s.velMultiplier += vel1;

In DyTGSContactPrep.cpp.

s.velMultiplier should be:

if(isKinematic0)
			s.velTarget -= vel0;
		if(isKinematic1)
			s.velTarget += vel1;

Can you make that change and see if that resolves the issue?

mottosso, 2020-10-28:
Thank you @kstorey-nvidia, having a look at these now.

alanjfs, 2020-10-28:
Thank you @kstorey-nvidia, having a look at these now.

alanjfs, 2020-10-28:
I can't quite nail down a consistent behavior. I can confirm that with a proper inertia tensor it is much more stable; I wasn't able to reproduce the case of it not being set; all rigids go through the a call to `PxRigidBodyExt::setMassAndUpdateInertia`. I suspected it just wasn't saved out with the PVD session, but that wasn't it. Will keep en eye on that moving forwards.

The issue with kinematics is still there, both with and without the bugfix you provided - no difference (I compared them both side-by-side, they produce exactly the same values).

Here is a clearer example of it, with a very weak `PxSpring.stiffness = 10`.

[madmusce2.zip](https://github.com/NVIDIAGameWorks/PhysX/files/5452498/madmusce2.zip)

[image]

It somehow inherits orientation from the kinematic rigid, which is the `actor0`. At higher strengths, aside from jitter which I'm also unsure of how to reduce, you can still see how it appears to follow the kinematic rigid to some extent.

[image]

At this point I would make due with PGS, if only I could get rid of the jittering. Is there a more suitable constraint/joint for this purpose?

kstorey-nvidia, 2020-10-28:
Thanks. That's the usual case (kinematic is first actor), which is why this typo slipped through.

If you can provide a repro for this case in a simple snippet, hopefully we can get to the bottom of it.

alanjfs, 2020-10-29:
I had a look at making a minimal repro outside of the overall project I'm working on, but it would be challenging. Will work around this for now but return here with a repro if it becomes a showstopper. Thanks @kstorey-nvidia

alanjfs, 2021-03-26:
It's come back to haunt me. :S This time in a simpler form.

[twoboxesbug.zip](https://github.com/NVIDIAGameWorks/PhysX/files/6214172/twoboxesbug.zip)

[image]

Like before, the problem appears *mostly* with TGS, with PGS it mostly behaves well.

[image]

Interestingly, this also happens with drives.

[twoboxes_drivebug.zip](https://github.com/NVIDIAGameWorks/PhysX/files/6214209/twoboxes_drivebug.zip)

[image]

I think I'll be able to put together a reproducible in this case, but if someone recognises this, let me know please!

alanjfs, 2021-03-27:
The problem is made much worse with a rotating kinematic body.

[image]
[image]

alanjfs, 2021-03-27:
Ok, here we go.

- https://gist.github.com/alanjfs/76416c401ca471d34da920190f6c43b2
- [twoboxes_reproducible.zip](https://github.com/NVIDIAGameWorks/PhysX/files/6215428/twoboxes_reproducible.zip)

Lines of interest are..

- [Damping settings for the joint](https://gist.github.com/alanjfs/76416c401ca471d34da920190f6c43b2#file-physxbug_reproducible-cpp-L41)
- [Choice between PGS and TGS](https://gist.github.com/alanjfs/76416c401ca471d34da920190f6c43b2#file-physxbug_reproducible-cpp-L66)
- [actor0 being kinematic and animated](https://gist.github.com/alanjfs/76416c401ca471d34da920190f6c43b2#file-physxbug_reproducible-cpp-L100)

And the result as seen from PVD, where TGS appears to introduce additional velocity to the dynamic rigid when jointed to an animated kinematic rigid, whereas PGS does not.

[image]
[image]

@kstorey-nvidia May I summon you?

alanjfs, 2021-03-27:
The same behavior can also be seen with the `PxDistanceJoint`.

```cpp
PxJoint* createJoint(PxRigidActor* parent, const PxTransform& parentTm,
                     PxRigidActor* child, const PxTransform& childTm) {

    PxDistanceJoint* joint = PxDistanceJointCreate(*gPhysics, parent, parentTm, child, childTm);
    joint->setDistanceJointFlag(PxDistanceJointFlag::eMAX_DISTANCE_ENABLED, true);
    joint->setDistanceJointFlag(PxDistanceJointFlag::eSPRING_ENABLED, true);
    joint->setMaxDistance(0.0f);
    joint->setStiffness(100.f);
    joint->setDamping(100.f);

    return joint;
}
```

[image]