NVIDIAGameWorks/PhysX #359: Issue with eREQUIRE_RW_LOCK flag and onContactModify(...) callback

Maclodawn, 2020-11-06:
Hello everyone,

In our project we are enabling the scene flag `PxSceneFlag::eREQUIRE_RW_LOCK` at initialization, which requires us to do a `void NpScene::lockWrite()` on the scene before calling `void NpScene::simulate(...)` and `bool NpScene::fetchResults(...)` followed by an `void NpScene::unlockWrite()`.

We are also using `void PxContactModifyCallback::onContactModify(...)` callback, which is required to be thread safe.

Now if we want to be able to read (not write) data from the actors given through `onContactModify(...)` we need to do a `void NpScene::lockRead()` on the scene or we would get an error.

Our issue is that since `onContactModify(...)` calls can be made on other threads than the main thread we cannot take the scene lock since it is already taken to be able to call `simulate(...)` and `fetchResults(...)`. This means that we are forced to either keep the error and ignore it, or disable the scene flag `eREQUIRE_RW_LOCK` which is helping us ensuring we didn't miss any lock when accessing the scene.

It seems to me there's a design issue. Am I missing something ?

Thanks for your help

kstorey-nvidia, 2020-11-06:
You only need to hold the write lock when calling simulate() and fetchResults() respectively. As soon as the call to simulate() has returned, you can release the write lock on the scene while the simulation executes asynchronously, allowing your callbacks to acquire locks if required.

Maclodawn, 2020-11-09:
Indeed, we were doing a blocking `fetchResults()` which was keeping the lock. So instead we now do loop checking `fetchResults()`' result, and it fixed our issue.
Thanks a lot for your fast answer !

closed 2020-11-09 by Maclodawn

kstorey-nvidia, 2020-11-09:
You're right that blocking with fetchResults(true) requires the write lock to be held. In this case, there is the option of calling checkResults(true) before acquiring the lock and calling fetchResults(), which will ensure that the lock is not acquired until the simulation has completed.  As an alternative, you can provide a continuation task for PhysX to execute after the simulation has completed.

Maclodawn, 2020-11-09:
Ah right, I forgot about that `checkResults()`, that's better !
In our case we do need to block the main thread so this solution is perfect for us.
Thanks.