NVIDIAGameWorks/PhysX #360: Set MaxImpulse on a contact constraint in onContactModify

spanzeri, 2020-11-10:
Hello,
I have been trying to use setMaxImpulse during an onContactModify callback in physx 4.1.
My understanding, reading the documentation, is that max impulse should limit the impulse that can be applied to resolve a contact constraint.
However, the impulse I receive in the onContact is the same as if I didn't set a maximum.
I was wondering if I misunderstood the documentation or if there is one or more flags I missed.

What my code does is:

In the filter shader I applied all the flags I believe are needed:
```c++
pairFlags = physx::PxPairFlag::eCONTACT_DEFAULT;
pairFlags |= physx::PxPairFlag::eDETECT_CCD_CONTACT;
pairFlags |= physx::PxPairFlag::eNOTIFY_CONTACT_POINTS;
pairFlags |= physx::PxPairFlag::eMODIFY_CONTACTS;
```

The onContactModify callback is called and I set the max impulse to a really low value (for testing purposes).
```c++
void ContactModifyCallback::onContactModify(physx::PxContactModifyPair* const pairs, physx::PxU32 count)
{
    for (size_t i = 0; i < count; ++i)
    {
        physx::PxContactSet& contacts = pairs[i].contacts;
        u32 nbContacts = contacts.size();
        for (u32 j = 0; j < nbContacts; ++j)
        contacts.setMaxImpulse(j, 0.1f);
    }
}
```

The onContact function contains the following:
```c++
void PHYSX_Manager::onContact(const physx::PxContactPairHeader& pairHeader, 
                              const physx::PxContactPair* pairs,
                              physx::PxU32 nbPairs)
{
    for (u32 i = 0; i < nbPairs; i++) {
        const physx::PxContactPair &cp = pairs[i];
        physx::PxContactStreamIterator iter(cp.contactPatches, cp.contactPoints, cp.getInternalFaceIndices(), cp.patchCount, cp.contactCount);
         physx::PxU32 hasImpulses  = (cp.flags & physx::PxContactPairFlag::eINTERNAL_HAS_IMPULSES);
         // ...
        while (iter.hasNextPatch())
        {
            iter.nextPatch();
            while (iter.hasNextContact())
            {
                iter.nextContact();
                physx::PxVec3 impulse = hasImpulses ? normal * impulses[nbContacts] : physx::PxVec3(0.f);
                nbContacts++;
                float maxImpulse = iter.getMaxImpulse;
                // ...
            }
    }
}
```
The maxImpulse value is the one I set during onContactModify.
However, the impulse.magnitude() value is the same as if the max impulse was not set and it's a much higher values.
Thanks in advance for your help.

kstorey-nvidia, 2020-11-11:
Thanks for reporting. Can you confirm whether you are using the PGS or TGS solver? What are you simulating? Standard rigid bodies or articulations?

zemlifr, 2020-11-11:
https://github.com/NVIDIAGameWorks/PhysX/issues/231 I have same problem. Works fine with PGS.

spanzeri, 2020-11-11:
I am using TGS and simulating rigid bodies only for now.
I tried @zemlifr suggestion and switched to the PGS solver and I can confirm it works in that case.

spanzeri, 2020-11-16:
I gave it a go to see if I could get TGS and maxImpulse to work.
I am not yet very familiar with the internals, so this might be non-sense, but it seems to work.

In DySolverConstraint1DStep.h
```c++
struct SolverContactPointStep
{
    PxVec3 raXnI;
    PxF32 separation;
    PxVec3 rbXnI;
    PxF32 velMultiplier;
    PxF32 targetVelocity;
    PxF32 biasCoefficient;
    PxF32 maxImpulse;  // <- Added
    PxU32 pad[1];          // Removed 4 bytes from the padding
};
```
In DyTGSContactPrep.cpp
```c++
static FloatV constructContactConstraintStep(/*...*/)
{
    // ...

    solverContact.maxImpulse = contact.maxImpulse;

    return penetration;
}

static FloatV solveDynamicContactsStep(/*...*/)
{
    // ...

        const FloatV appliedForce = FLoad(forceBuffer[i]);

        // Removed: const FloatV maxImpulse = FMax(); //KS - todo - hook up!
        const FloatV maxImpulse = FLoad(c.maxImpulse);

        //Compute the normal velocity of the constraint.
        const Vec3V v0 = V3MulAdd(linVel0, contactNormal, V3Mul(angState0, raXnI));
    // ....
}
```

kstorey-nvidia, 2020-11-17:
Thanks for making this effort. Your implementation is on the right tracks. However this is not a complete solution. There is also the  block constraint model that needs updating. The block constraint model solves 4 independent contact constraint at a time using SIMD. The code you modified is used to solve singular contact constraints.

You would also need to add the equivalent code to DyTGSContactPrepBlock.cpp.

It should be equally easy to add. Apologies for the omission. This will be included in the next release.

spanzeri, 2020-11-17:
Thanks for the tip Kier. I had a look at the DyTGSContactPrepBlock.cpp, but it looks to me like the max impulse is implemented for that path which is somewhat surprising. I wonder if I missed something.
The value is stored at line 784, inside setupFinalizeSolverConstraints4Step.
```c++
if (hasMaxImpulse)
{
    maxImpulse[contactCount - 1] = V4Merge(FLoad(con0.maxImpulse), FLoad(con1.maxImpulse), FLoad(con2.maxImpulse),
        FLoad(con3.maxImpulse));
}
```
In solveContact4_Block it is read at line 2552
```c++
if (hasMaxImpulse) 
{
    maxImpulseMask = 0xFFFFFFFF;
    maxImpulses = reinterpret_cast<Vec4V*>(currPtr);
    currPtr += sizeof(Vec4V) * numNormalConstr;
}
```
and used at line 2649.
```c++
const Vec4V newAppliedForce = V4Min(V4Add(appliedForce, _deltaF), maxImpulse);
```
I tried a quick test with 6 boxes with max impulse and I get both the values and results I expect (and I verified I am going through the 4_block implementation).
Did I miss something else?
Thanks a lot for the help

kstorey-nvidia, 2020-11-17:
Actually, you are right. Sorry, I wrote this assuming that it would be missing in both places without verifying. I also just verified that this is correctly included in the GPU solver so it is only missing in the piece of code you included in your response.

spanzeri, 2020-11-17:
No problem at all, I am happy I could figure out how to do it.
I'll be happy to create a pull request if you'd like, or if it's better for you I'll use my locally patched version and wait for the next release and close the issue as resolved