NVIDIAGameWorks/PhysX #362: Custom joints

YashaAkimov, 2020-11-18:
for example: i want to create cam, wich transforming rotary motion into linear motion. 
[image]
its wery easy only if i need transfer motion from bottom to top, just using sweeps. But its not consider forces, applying to top side of cam.
another way is implement full mechanism in simulation world, but I think it wold have poor performance, because its need to calculate collision between two or more high poly objects and solve at least 3 standart joints instead of solving one custom joint.
So it's possible to create joint based on custom math model ?

kstorey-nvidia, 2020-11-18:
Yes, it is. Take a look at "SnippetCustomJoint" in the PhysX SDK. That snippet demonstrates creating a pulley joint as a custom joint.

YashaAkimov, 2020-11-18:
Many thanks.

closed 2020-11-18 by YashaAkimov