NVIDIAGameWorks/PhysX #363: isValidFloatV() can divide by zero

fredUmlaut, 2020-11-19:
in PsWindowsInlineAoS.h
PX_FORCE_INLINE bool isValidFloatV(const FloatV a)
`if (			(PxAbs((x - y) / x) < FLOAT_COMPONENTS_EQUAL_THRESHOLD) && ..`

If x is 0 then a divide by zero will happen.

kstorey-nvidia, 2020-11-23:
FloatV has an earlier test that checks if the difference in abs value between x and y, z and w is below FLOAT_COMPONENTS_EQUAL_THRESHOLD.

In the case of x being 0, it should return in this case unless the FloatV itself is not value (all elements should be the same).

The divide by zero shouldn't cause problems because FP exceptions are disabled in PhysX. However, it should indicate that the FloatV that was supplied is not a legal FloatV. Do you have an example that is triggering this to fail? You should see the validation code assert in this case.

fredUmlaut, 2020-12-08:
We had no actual case but our compilers display warnings that a division by zero can happen.

closed 2021-04-22 by kstorey-nvidia