NVIDIAGameWorks/PhysX #365: D6 joints jitter on ragdoll

Esentiel, 2020-11-20:
Hello.

We have ragdoll implementation using D6 joints with soft limits. It works fine on high fps, simulation rate.
Sadly it doesn't work well on low fps(20-30) and low simulation rate(~30). Joints produce jitter. [We use PGS in the project but switching to TGS didn't help]

On the other hand there is no such issues on fps and simulation rate around 100.
Expamle:
[image]

So what I tried:
1. enabled projection for joints (didn't help)
2. tweaked max depenetration vel, contact/rest offset (didn't help)
3. relaxes joints limits (didn't help)
4. tweaked mass of the bodies. this helped in most cases but ragdoll was looking not physically/visually correct. Artists didn't agree to this solution.

I tried to make use of PVD but wasn't able to collect all the necessary information about joint.
What I was able to collect you can find in the link below(.pxd2): https://drive.google.com/file/d/1SC5bFUe-6WemCRKsqEG8vfZUbovOPZnb/view?usp=sharing

Also I gave it a try to debug PhysX code but it appears to be pretty complicated part of code related to constraints/contacts, so I gave up on this for now. If you can point me to spots in code where I can start digging would be just awesome.

Based on what I've tried and what I've seen, my guess that joint's limits can't be solved by a solver that's why bodies remains awaken for a while trying to solve the constraints. But for some reasons we can't observe such behavior on the higher simulation rate. maybe there is some clamping for say impulses that on higher simulation dt clamp necessary impulse to push other bodies and fulfill the constraint. Nevertheless I wasn't able to fins anything like that.

Any advice is highly appreciated!

kstorey-nvidia, 2020-11-20:
One observation I can make from a quick look is that your contact offset is 0.01 and your rest offset is 0.0095. That is a far smaller range than can probably be simulated reliably at larger time-steps.

Contact offset defines the distance at which we start generating contacts. Rest offset is how much we subtract from the reported distance between the shapes inside the solver, which can be used to make objects either remain slightly separated or slightly penetrated.

Can you try increasing contact offset to something larger, e.g. try 0.05, while leaving rest offset as it is to see if that causes any improvements?

The problem with having contactOffset - restOffset be very small is that you give the solver a very narrow window to work within. 

In this case, there is a 5mm margin between the state the body should be at rest and the state in which the contact generation decides that the shapes are disjoint. If the solver overshoots by more than 5mm (as it can if there's some deep penetration), the contact generation will reject the pair. Then gravity will cause the objects to fall back into deep penetration, causing jitter.

If the contact gen generates distant contacts, the solver can use these to ensure that the shapes don't end up penetrated at the end of the frame.

As you noted, this works fine with smaller time-steps because gravity moves the shapes much less in a single frame.

The gap between contact distance and rest distance needs to be significantly larger than the distance the object will move under gravity in a single frame. At 60hz, that distance is around 2.7mm using Euler integration (vel = 9.8*1/60, position delta = vel * 1/60). At 30hz, that distance is around 1.1cm (vel = 9.8*1/30, position delta = vel * 1/30).

If this doesn't help, can you resend a capture with:

pvdClient->setScenePvdFlag(PxPvdSceneFlag::eTRANSMIT_CONSTRAINTS, true);

set to true so we get the constraints through in PVD? This might help diagnose any issues with the joint configurations.

Pierre-Terdiman, 2020-11-20:
Similarly, it's a good idea to increase the contact distance parameter of joint limits as much as it can be.

Esentiel, 2020-11-20:
Thank you for a fast response! I tried to tweak contactOffset, restOffset and contact distance as you suggested, and maybe it became slightly better, nevertheless jitter is still a case.

I would like to send constraint information using PVD but it's not possible for now. I enabled PxPvdSceneFlag::eTRANSMIT_CONSTRAINTS and it only shows limit cones, and no any entities under "Joints" list.

[image]

I tried to compare our code/visualization settings to SnippetJoint project from PhysX code snippets and it seems, SnippetJoint produces joint information even with PxPvdSceneFlag::eTRANSMIT_CONSTRAINTS disabled. Also I wasn't able to find any code that enable PxConstraintFlag::eVISUALIZATION in SnippetJoint, but it still collects and shows constraints information under Joints list in left-side menu. Which seems odd to me. 

[image]

So I need to take a small break to find out how to send constraint info into PVD for our project. Otherwise I can't provide you with necessary information unfortunately.

Esentiel, 2020-11-20:
Form what I see in debugger, somehow in PhysXExtensions we hit this version of updatePvdProperties():
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/physxextensions/src/ExtJoint.h#L121

Which is so strange because PX_SUPPORT_PVD is enabled through all the solution and PhysX libraries, but it seems except PhysXExtensions (or ExtJoint.h). Wil try to dig into the issue on the next week.

kstorey-nvidia, 2020-11-20:
Did you call PxInitExtensions(...) and did you pass the PVD instance to PxInitExtensions(...)?

Esentiel, 2020-11-23:
Yes, I did that. It seems my issued was in linking PhysXExtensions in a wrong way. Now I fixed that.
Here is new info: https://drive.google.com/file/d/1rq9aDIBolFidPqOrDILmBZxD3SLci61y/view?usp=sharing

kstorey-nvidia, 2020-11-23:
Thanks. This capture seems to settle pretty quickly. Do you have an example capture of it jittering.

I can echo what Pierre suggested - you might be able to increase the contact distance on the joints to improve limit handling. We might be seeing the same issue as we saw with contacts, where the joint is bouncing into/out of a limit error state rather than being held stably at the limit.

It may also be a good idea to try turning off the sleeping on the ragdoll by either setting the wake counter to be some arbitrary large value or by setting the sleep threshold to be 0. With this capture, it is possible that there might have been a small amount of jitter but the ragdoll fell asleep almost instantly so it is hard to tell if that would have been a problem.

Esentiel, 2020-11-23:
I beg your pardon. I streamed data via network into PVD and I had 1500 frames there. When I saved it I got what I posted, very tiny part of the real simulation for any reason.
New file can be found here - https://drive.google.com/file/d/1NPtYPQ6-bj6Vth3RF1HDiPfCM2iktnfF/view?usp=sharing

Regarding joint contact distance: initially in our project it was always set to 0. I changed it to default value (-1) so sdk can calculate the actual value that obey all the limitations according to this code snippet
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/include/extensions/PxJointLimit.h#L320

Regarding to sleep count, I assume it's some kind of workaround - to force the dynamic parts avoid simulation. I would like to try fix this behavior via standard methods that SDK provides like multiplier and some settings. If I fail at this, I will have no other options than that :-)

Esentiel, 2020-11-25:
Hey there!

I re-tested everything once more. And now I can see, that we have two distinct issue:

1. Ragdolls jitter after activation till they fall asleep [Almost 100% reproduce on low simulation rate]
2. In some poses, some constraints can't be solved and produce jitter infinitely [I was able to reproduce on 60hz, but reproduction rate is much lower]

Thanks to @kstorey-nvidia suggestion/explanation regarding contact offset and rest offset we were able to fix point 1 from the above list. Now there is no jitter after ragdoll activation till it fall asleep.
Point 2 wasn't solved using adjustments of joint's contact distance unfortunately. I tuned mass of ragdoll shapes and when mass is equal for every part - there is no reproduce for point2 issue, but it leads to non-human visual. So I'm thinking to increase sleep threshold, this helped to force ragdoll to fall asleep, thus jitter will no be the case anymore.

kstorey-nvidia, 2020-11-25:
If you haven't already tried this, please try increasing the number of solver position iterations to see if it helps with (2). Also, consider if you can do things like add damping to the joints' limited/free axes. This will resist motion around degrees of freedom. 

There is a possibility that there may be cases that cannot be physically solved at all, e.g. a contact between shapes that cannot be separated due to joints. If these cases occur, the only real solution would be to filter those collisions.

Esentiel, 2020-11-25:
I tried increasing solver iterations for position from 4 to 20 and didn't notice any difference in visual. Also I tried increasing damping on joints, same - no difference, unfortunately.

By filtering contacts you mean that we can use onContactModify() function to disable some contacts there? Sounds like a clean solution, but I'm not sure how to find the contacts that produce jitter, will dig into this, thanks!

kstorey-nvidia, 2020-11-25:
You don't necessarily need to use contact modification. I was thinking the simulation filter shaders, which culls pairs at the time the shapes begin to interact so they are never processed by contact generation.

Culling in contact modification is less efficient because we do all the contact gen work, then you tell us to throw them away.

Esentiel, 2020-11-26:
I'm trying to take a look from another point of view.

I made capture from the same situation but with 200hz and noticed that on 200hz there are now blue arrows that show velocity, only grey ones. I assume blue one shows impulse after penetrating between shapes? I increased contact offset to 0.25 and still have these blue velocity arrows.
link to pvd data - https://drive.google.com/file/d/16dz8QKwDqD9g405nJtOLpknWF_7Qv6Vm/view?usp=sharing
Probably that's something that can't be avoided on low simulation rate. So it's time to implement workaround :|

I want to thank you! you helped me a lot with this issue, appreciate so great support from PhysX team!

closed 2020-12-02 by Esentiel