NVIDIAGameWorks/PhysX #371: PhysX 5.0 - Available in 2020 ？

swq0553, 2020-12-29:
PhysX 5.0 is just around the corner, and we wanted to provide a look at all the new features! In this version, available in 2020, we’ll be introducing support for a unified constrained particle simulation frame。

Tody is 2020.12.29！

Zeblote, 2020-12-29:
PhysX 5.0 is old news already, when will you release PhysX 5.1? Or is it not going to be public anymore now that Epic stopped sponsoring the open source release?

amoravanszky, 2021-01-08:
Hi guys, 

on this topic I recently posted this on the NVIDIA forums:

Omniverse has released into Open Beta in Dec 2020, with various Omniverse applications built on PhysX 5.  Earlier this year Isaac Sim has launched on PhysX 5.  

https://www.nvidia.com/en-us/design-visualization/omniverse/

To this you will no doubt reply "but I wanted a C++ SDK just like before".  The answer to that is, the C++ SDK is not done.  You mention Epic above.  My team has internally relied on UE PhysX for our own needs, but as a result of their change in strategy we need to also find a different solution to make our own content.  This means that the PhysX team has to now develop all of the physics content creation tools ourselves within Omniverse.  One one hand this is a good thing because it will make for a seamless experience and our team will be much more intimately familiar with real content, on the other hand it means that our previous promised SDK roadmap has completely changed.

I can't promise a new date of availability for the PhysX 5 SDK, all I can say is we're still working on it, and it will be released when you and I are happy with the fullstack of content creation tools on top of it in Omniverse.

StlGameTesters, 2021-01-08:
In case you need to forward this message again:  (duplicate word)

“One one hand this is a good thing because it will make”

Hi guys,

https://www.nvidia.com/en-us/design-visualization/omniverse/

Zeblote, 2021-01-08:
> and it will be released when you and I are happy with the fullstack of content creation tools on top of it in Omniverse.

But most users of the PhysX SDK don't actually need any of those things. They already have PhysX integrated in a proprietary engine with an estabilished content pipeline, or are interested in upgrading the outdated PhysX integration in Unreal Engine, etc. Where do you see custom content tools (which would be incompatible with these situations) as being useful or required?

I just think it's sad that all the improvements and optimizations that have accumulated in the almost two years (!) since PhysX 4.1 are being held back indefinitely...

closed 2021-01-25 by swq0553

Hoikas, 2021-02-02:
An additional shame is the multitude of out of tree patches that have to be applied to build the public version of PhysX on modern compilers. By the time a new SDK is released, there will be little interest because these problems will have caused everyone to move on to another physics middleware.

Zeblote, 2021-02-02:
Implies that another physics engine capable of competing with PhysX actually exists. What would you use instead?

Hoikas, 2021-02-02:
Seems like there are plenty of options to me, depending on your workload:
* Havok
* Bullet
* NewtonGameDynamics
* or, if you really want to stab your eyes out... ODE

It seems like NVIDIA engineers are trying to compete with Unreal Chaos instead of responsibly curating the community and product they already have. This might be acceptable if the development were happening in public, the product was already released, or if the current PhysX 4.1 branch actually compiled on modern systems. IMO, it sends a very bad message to users when your public-facing code has compile issues that have existed for years and yet the open pull requests on these issues have been either ignored or closed but not merged back into the GitHub code.

Zeblote, 2021-02-02:
Havok is ridiculously expensive, so that's not an option.
Bullet, Newton, ODE, are all far slower than PhysX.
Unreal Chaos is currently a complete disaster performance wise.

There really aren't any good competitors.

Hoikas, 2021-02-02:
That is your personal opinion... plenty of products, some free, some paid, ship with those physics engines.

DomiOh, 2021-02-14:
We used Bullet for our engine. It is way too slow. Even raycast tests are hillariously slow. It is not useful for game development. Newton is not even able to compete with PhysX and ODE is too low level. I agree to Zeblote, there is no good alternative to physix (yet).

DaveGravel, 2021-02-15:
I like PhysX 4.1, It is a good engine and fun to use. 

About Newton it is false to say it is slow.
It is false too when you say it can't compete with PhysX. 
Newton can compete with PhysX on a lot simulations, 
Not only compete but in some special cases Newton can do better.
Newton is very easy to use too with precision, It is a lot harder to get this quality from PhysX.

PhysX GPU is very fast but a lot more limited on the physics quality and shape precision.
When you need really good precision with PhysX you need to use special world setup and it have a big speed cost.
In this case PhysX is not really faster that Newton, With Newton the default setup already have very good precision.

The new Newton 4 is very promising and the progression going pretty fast too.
For my part I think PhysX and Newton is both very nice engines and both is the better solutions that you can find today.
Sorry about my english to explain things.

Zeblote, 2021-03-31:
So I guess 2 more weeks until this mysterious GTC talk?

Zeblote, 2021-04-12:
@amoravanszky I just watched the GTC presentation, there is some very nice stuff there, especially excited about the support for user shapes with custom collision detection. But there was no mention at all on when we might see the PhysX 5.1 SDK release...

alanjfs, 2021-04-13:
> I am now preparing a GTC talk where I hope to shed more light on what exactly is going on and what we've been up to.

The time is now upon us. :) Will a certain someone announce a certain something? Is there a calendar date and time to look forward to?

alanjfs, 2021-04-13:
> @amoravanszky I just watched the GTC presentation, there is some very nice stuff there, especially excited about the support for user shapes with custom collision detection. But there was no mention at all on when we might see the PhysX 5.1 SDK release...

Oh haha, I missed this. Where is this GTC presentation? There are so many of them, I wasn't able to spot anything involving PhysX. :S @Zeblote Do you have a link?

Zeblote, 2021-04-13:
@alanjfs https://gtc21.event.nvidia.com/media/Omniverse%20Physics%20%5BS31840%5D/1_3qfecbz5

Zeblote, 2021-04-19:
One could use PhysX to create a really realistic simulation of tumbleweeds rolling past.

tlf30, 2021-05-20:
Any update on PhysX 5 SDK availability?  
Just wondering if we should still expect to see PhysX released as a standalone SDK...

Zeblote, 2021-05-20:
Very unlikely. It's in their best interest to keep it exclusive to that Omniverse thing. If only that didn't exist...

LaserYGD, 2021-11-15:
Still nothing I assume? 
I was hoping unity would upgrade to Physx 5, but we get Omniverse, which is useless to me

xCyborg, 2022-02-23:
Any news of PhysX 5.0 SDK?

SummerInLake, 2022-03-10:
Chaos has many problems and doesn't meet my needs. I still hope to use PhysX in ue in the future. I hope they don't give up

mastercoms, 2022-03-10:
Same, PhysX 4 in UE has been useful (not without its pains of independent integration) but PhysX 5 will help a lot!

nickolay411, 2022-03-23:
@amoravanszky  How is the Physx 5.0 SDK coming along?  Can you give us an update, even a small one to help shed some light on what's happening? Thanks!

duplexsystem, 2022-10-18:
Given that it's been pretty much radio silence it's likely that we might never see a PhysX 5.0 SDK release. We have to remember that Nvidia doesn't owe us a PhysX 5.0 SDK release but still sucks nonetheless. I hope Nvidia proves me wrong.

ErikSom, 2022-10-29:
Has anyone checked out https://github.com/jrouwe/JoltPhysics , powered the PS5 Horizon game physics. Looks pretty awesome, but no fluids..

DanielGibson, 2022-11-08:
looks like it's finally released and lives in a different repo now:
https://developer.nvidia.com/blog/open-source-simulation-expands-with-nvidia-physx-5-release/
https://github.com/NVIDIA-Omniverse/PhysX