NVIDIAGameWorks/PhysX #375: Can't get articulation joint drive working, what am I doing wrong?

FredrikNoren, 2021-01-06:
Hi,

I'm trying to set a drive velocity to an articulation joint, but some reason it doesn't seem to apply any force and I can't figure out why. The articulation itself seems to work fine (the objects are connected with the joint, it's dropped from a height and moves naturally), and I can get a drive to work if I use Joints instead. Here's my code (it's rust but it should map pretty much 1-1 to cpp):

```rust
let physics_material = Material::new(&physx.physics, 0.5, 0.5, 0.6);
let root = {
    let articulation = Articulation::new(&physx.physics);
    let link = ArticulationLink::new(&articulation, None, &Transform::from_translation(vec3(0., 0., 10.)));
    let shape = Shape::new(&physx.physics, &BoxGeometry::new(1., 1., 1.), &[&physics_material], true, ShapeFlags::default());
    link.attach_shape(&shape);
    link.update_mass_and_inertia(10., None, false);
    self.world.create(EntityBuilder::new()
        .set(MESH, gpu.box_mesh.clone())
        .set(MATERIAL, self.material_yellow)
        .set(ARTICULATION_REDUCED_COORDINATE, articulation)
        .set(ARTICULATION_LINK, link)
        .set_default(TRANSLATION)
        .set_default(ROTATION)
        .set_default(LOCAL_TO_WORLD)
    )
};
let arm = {
    let articulation = self.world.get_ref(root, ARTICULATION_REDUCED_COORDINATE).unwrap();
    let parent_link = self.world.get_ref(root, ARTICULATION_LINK).unwrap();
    let link = ArticulationLink::new(articulation, Some(parent_link), &Transform::from_translation(vec3(2., 0.0, 0.)));
    let shape = Shape::new(&physx.physics, &BoxGeometry::new(1., 1., 1.), &[&physics_material], true, ShapeFlags::default());
    link.attach_shape(&shape);
    link.update_mass_and_inertia(10., None, false);
    let joint = link.get_inbound_joint();
    joint.set_parent_pose(&Transform::from_translation(vec3(-1., 0., 0.)));
    joint.set_child_pose(&Transform::from_translation(vec3(1., 0., 0.)));
    joint.set_joint_type(ArticulationJointType::Revolute);
    joint.set_motion(ArticulationAxis::Twist, ArticulationMotion::Free);
    joint.set_drive(ArticulationAxis::Twist, 10., 0.0, f32::MAX, ArticulationDriveType::Force);
    joint.set_drive_velocity(ArticulationAxis::Twist, 3.);
    self.world.create(EntityBuilder::new()
        .set(MESH, gpu.box_mesh.clone())
        .set(MATERIAL, self.material_yellow)
        .set(ARTICULATION_LINK, link)
        .set_default(TRANSLATION)
        .set_default(ROTATION)
        .set_default(LOCAL_TO_WORLD)
    )
};
```

preist-nvidia, 2021-01-06:
`    joint.set_drive(ArticulationAxis::Twist, 10., 0.0, f32::MAX, ArticulationDriveType::Force);
`
The issue seems to be that you are setting the drive damping to 0.0 - the damping will act on the velocity error while the stiffness acts on the position error. See if a nonzero value helps.

FredrikNoren, 2021-01-06:
@preist-nvidia I tried 0.01, 0.1, 1.0 and 10.0, but none of them seems to change anything. It just looks like the drive isn't doing anything.

preist-nvidia, 2021-01-06:
I should have mentioned that you should set the stiffness to zero, otherwise the position control part of the drive will fight the velocity drive.

FredrikNoren, 2021-01-06:
@preist-nvidia Hm still nothing. I tried stiffness=0.0 and damping=0.1 and 1.0

preist-nvidia, 2021-01-06:
What if you set damping to a huge value? Is 1.0 damping a realistic value to be able to overcome gravity given the link's mass?

FredrikNoren, 2021-01-06:
@preist-nvidia That works! I think I tried a big value on stiffness at some point but not damping. Thanks a bunch!

closed 2021-01-06 by FredrikNoren