NVIDIAGameWorks/PhysX #377: Unsure about weird axes behavior when attaching a second shape to a Character Controller

aegooby, 2021-01-11:
So I have a character controller I created using `PxPhysics::createController`, and in the `desc` I specified the `upDirection` field as `PxVec3(0, 0, 1)`, so I want the z-axis to be the vertical one. This works perfectly, giving me a hitbox around my character that looks like this:
[image]
Now, I want to create an additional trigger shape used for the player to attack, which should be in front of the player. When I created a shape and used `setLocalPose(PxTransform(PxVec3(100, 0, 0)))`, I ended up with the shape somehow being transformed up (in the z-direction) instead of where I wanted it to go:
[image]
I would have thought it would move sideways in some direction, but definitely not up! Am I misunderstanding something about how `setLocalPose` works?

Pierre-Terdiman, 2021-01-12:
If you could share some code, that might help.

One important thing to know here is that PhysX capsules are oriented along the X axis by default. So if you create a capsule with an identity pose, it is going to be X-aligned. In order to align the capsule with either Y or Z (the traditional up directions), you need to rotate that initial X-aligned capsule, using either the capsule's global pose, or its local pose. In the Character Controller module I think it was achieved using the capsule's global pose (in Controller::createProxyActor). This means that this rotation will also affect any extra shape you add to the actor - it would have been different if we would have used the capsule's local pose for aligning it to Z, instead of its global pose.

Thus, you might need to cancel this actor / global rotation in your extra shape's local pose, i.e. instead of:

setLocalPose(PxTransform(PxVec3(100, 0, 0)))

you might need:

setLocalPose(PxTransform(PxVec3(100, 0, 0), Rot)),

with Rot a quaternion opposed to the one used in Controller::createProxyActor. The quat in that function was initially computed like this:

	const PxQuat q = PxShortestRotation(PxVec3(1.0f, 0.0f, 0.0f), up);

i.e. it's a rotation that transforms the capsule's default X axis (1;0;0) to the user-defined up vector (0;0;1) in your case.

So, try this:

// This rotation goes from Z back to X, i.e. cancels the actor's rotation
const PxQuat q = PxShortestRotation(PxVec3(0.0f, 0.0f, 1.0f), PxVec3(1.0f, 0.0f, 0.0f));
setLocalPose(PxTransform(PxVec3(100, 0, 0), q)),

aegooby, 2021-01-13:
Okay, that worked great. Thanks for this. I had another question: somewhere else I saw an answer regarding character controllers that said they can only be made with kinematic actors containing one shape. Would it cause issues adding extra shapes? For example, if I want to have some kind of snake character with several segments, but have the head be controlled as a character controller, would that be possible? Also, is it possible to attach joints to the actors created by a character controller?

aegooby, 2021-01-13:
Update: here is some code that I was working with
```
PxControllerManager* controller_manager = ...;

// Create the controller
desc.radius = 50;
desc.height = 80;
...
desc.upDirection = PxVec3(0, 0, 1);

auto controller = controller_manager->createController(desc);
auto actor = controller->getActor();

// Add second shape
PxRigidActorExt::createExclusiveShape(*actor, PxBoxGeometry(...), *material);
```

Now interestingly, what I noticed is that when I applied the quaternion rotation and then the 100 x-axis position transform it still moved the box upwards. When I didn't apply it, the box moved upwards, only instead now it was rotated so that the z-axis was facing sideways rather than up. My workaround right now is to have a small matrix mapping the controller axes to my game axes, but this seems expensive and error-prone. I was wondering if there was another way to do this from within PhysX itself with the character controller so that setting `upDirection` as I had allows me to work with z as the upward axis.