NVIDIAGameWorks/PhysX #378: How to import model like FBX,OBJ,XML into Physx 4.1?

Timkeeper2018, 2021-01-13:
HI all,
I am a beginner. I want to import externally created models (OBJ, FBX, XML...) into PhysX, but all I can find are outdated codes. Does the current SDK include the snippet example for importing the model or what should I do to import the model and correctly implement the collision simulation?
Thanks a lot

kstorey-nvidia, 2021-01-13:
PhysX doesn't provide importers. The expectation is that this will have this as part of your larger application.

PhysX does provides cooking functionality to create PxConvexMesh and PxTriangleMesh from arbitrary source meshes. Cooking a convex mesh requires only the set of vertices to be provided. Cooking a triangle mesh requires a set of vertices and set of triangle indices. 

Other primitives like boxes, spheres, capsule, heightfields, planes require the application to define the properties of these geometries directly, e.g. radius, dimensions, sample points etc.

If you need to make a convex decomposition from a mesh, there are libraries like VHACD that you can use.

Timkeeper2018, 2021-01-13:
thanks a lot

closed 2021-01-13 by Timkeeper2018