NVIDIAGameWorks/PhysX #379: Missing px.h

NeoMorphed, 2021-01-13:
When I try to include "PhysX/PxPhysicsAPI.h" in my project, it gives me an error saying it cannot open foundation\px.h.  I took a look in the include files and it looks like there are a lot of missing files that PxPhysicsAPI.h is suppose to include such as:

#include "foundation/Px.h"
#include "foundation/PxAllocatorCallback.h"
#include "foundation/PxBitAndData.h"
#include "foundation/PxBounds3.h"
#include "foundation/PxErrorCallback.h"
#include "foundation/PxErrors.h"
#include "foundation/PxFlags.h"
#include "foundation/PxIntrinsics.h"
#include "foundation/PxIO.h"
#include "foundation/PxMat33.h"
#include "foundation/PxMat44.h"
#include "foundation/PxMath.h"
#include "foundation/PxPlane.h"
#include "foundation/PxPreprocessor.h"
#include "foundation/PxQuat.h"
#include "foundation/PxSimpleTypes.h"
#include "foundation/PxStrideIterator.h"
#include "foundation/PxTransform.h"
#include "foundation/PxUnionCast.h"
#include "foundation/PxVec2.h"
#include "foundation/PxVec3.h"
#include "foundation/PxVec4.h"

I am not sure exactly how to solve this problem because even the github include\foundation folder only includes:

PxAssert.h
PxFoundationConfig.h
PxMathUtils.h

Thanks

kstorey-nvidia, 2021-01-14:
Those files are in PxFoundation/include/foundation. The foundation library in PhysX is split into 2 parts: PxShared/include/foundation is a shared library used by several gameworks libraries while the files in physx/include/foundation are exclusively used by the PhysX library.

NeoMorphed, 2021-01-14:
Ahh, thank you!

closed 2021-01-14 by NeoMorphed