NVIDIAGameWorks/PhysX #381: How to improve dynamic character controller and kinematic platforms interaction

Esentiel, 2021-01-18:
Hello!

We use custom dynamic character controller in our project and we have some sort of kinematic platforms (say elevator or train etc.) on which player can move and which are moved in the world coordinates at the same time. Todays implementation isn't very great and it basically makes cct kinematic and implement very basic rules for it while on platform. I'm researching ways to improve this.

My first try was to simply add velocity of platform to the players velocity. this worked good enough on low velocities but I'm afraid when a platform will move fast enough it will produce issues with player collisions etc, so I assume this way isn't very safe.

Another option was to add say separate scene for platform and move player to that scene when needed. Idea behind that was that player doesn't get any velocity related to main scene(velocity of a kinematic platform itself). 
Unfortunately I wasn't able to find any way to move scene origin apart from origin of another scene. Is this even possible or maybe my use case for scenes is not covered in the current scenes approach?
Maybe there are some other ways to make friendship between dynamic character controller and kinematic platform? :)

Any help is appreciated!

Pierre-Terdiman, 2021-01-18:
SnippetNestedScene might be relevant for that second approach.

Otherwise I think people sometimes use a joint between the dynamic character controller and the world/platform.

For the kinematic character controller we have a specific piece of code managing this scenario, and the motion of the platform is explicitly tracked and added to the motion of the character. Something similar might be needed here as well - my experience with dynamic character controllers is limited.

Esentiel, 2021-01-18:
Thanks for a rapid response!

Yeah I've looked through standard PhysX kinematic character controller and it's way to manage kinematic platform interactions and it's pretty straightforward. First approach that I listed above (sum player's and platform's velocities) was inspired by the one that implemented in PhysX :) 
I checked SnippetNestedScene (to my shame I didn't find it before creating this ticket) and what it does is tracks delta velocity of "platform" in main scene and uses addForce() to apply this delta to actors inside the nested scene. From my point of view it's pretty much the same I was doing adding platform's velocity and player's velocity :)

Using joint between platform and player is very interesting idea, I didn't think of it before you mentioned. Will dig into this way, many thanks!

Esentiel, 2021-01-19:
I digged into nested scene a bit deeper(using PVD :) ) and now I realized that it's what I was looking for. Nested scene stays at the same spot in physical world, we only add position of truck to position of cubes to render them. Meanwhile cubes velocities are not related to trucks movement. Ideal solution from my point of view.

Thnaks!

closed 2021-01-19 by Esentiel