NVIDIAGameWorks/PhysX #383: How physx push away intersection shapes

echnyee, 2021-01-22:
Hello, there
I'm now confused about the way how physx push away intersection shapes.
Can anyone help to explain how the mechanism works.
Thanks in advance.
What I do is put 2 rigidbody(with one box shape) intersect with each other.
What I expect is these 2 boxes will be pushed away with some impulse, and depend on how deep they intersect, the impulse will be greater. so these 2 boxes will fly away
while what I see in pvd is these 2 boxes are just be pushed away a little and stops moving when they are not intersect.
Is this how the physx push away intersection shapes or there are some other factors will lead to this result.

[image]
[image]
[image]

kstorey-nvidia, 2021-01-22:
PhysX attempts to correct error without introducing energy because contacts should not add energy into a simulation. It does this by solving using position and velocity iterations. "Position iterations" solve constraints which includes a bias term intended to correct error. They are still solving on the velocity level, and this resulting velocity is used to update the pose of the rigid bodies. Following this, "velocity iterations" solve the same constraints, but with the biases from penetrations removed. These iterations make a best-effort attempt to remove any energy introduced by the position iterations to correct penetrations. The final result is the set of velocities that are carried over to the next frame of simulation. By default, PhysX runs 4 position iterations and 1 velocity iteration. 1 velocity iteration is usually only sufficient to remove a proportion of the energy introduced by error correction in the position iterations, but in some simple cases like the one you described, 1 is sufficient to correct all of the energy gain.

echnyee, 2021-01-25:
> PhysX attempts to correct error without introducing energy because contacts should not add energy into a simulation. It does this by solving using position and velocity iterations. "Position [...]

Great, thanks for your explaination