NVIDIAGameWorks/PhysX #384: Serialization of articulations is missing current joint state

mrTag, 2021-01-26:
My scene has reduced coordinate articulations with several revolute joints. For the game, we are making we need to save the whole physics state regularly so that the player can jump back to a previous state and try something different. So I serialize the whole physics state into a binary memory buffer like this:
```cpp
physx::PxSerializationRegistry* registry = physx::PxSerialization::createSerializationRegistry(*physXData.Physics);

physx::PxCollection* everythingCollection = physx::PxCollectionExt::createCollection(*physXData.Scene);
physx::PxSerialization::complete(*everythingCollection, *registry);

physx::PxSerialization::serializeCollectionToBinary(outStream, *everythingCollection, *registry);
everythingCollection->release();
registry->release();
```

and deserialize it like this:
```cpp
physx::PxSerializationRegistry* registry = physx::PxSerialization::createSerializationRegistry(*physXData.Physics);

physx::PxCollection* collection = physx::PxSerialization::createCollectionFromBinary(
		physXData.ActiveRestorePoint->GetAlignedDataPointer(), *registry);
physXData.Scene->addCollection(*collection);

collection->release();
registry->release();
```
And this works perfectly, all the objects, articulations and joints are there. The only problem is that the joints in the articulations are reset to their initial positions and zero velocity. The base of the articulation has the correct position and velocity, but the character basically resets to its T-Pose.
So my question is: Do I have to add anything to the serialization for the joints to also get their current state serialized?

The `PxCollectionExt::createCollection(PxScene& scene)` call I am using has a section for articulations, so I thought that will serialize the complete articulation state. I also had the `PxCollectionExt::createCollection(PxPhysics& physics)` call additionally in there at one point, but it didn't change anything, so I left it out. My guess is that the `PxSerialization::complete` call pulls in the remaining necessary data anyway.

I read something about the `PxArticulationCache` in the documentation, so I probably could save the cache of every articulation separately and apply that cache after deserialization. But I hoped the physx serialization would also save this information. Am I wrong here?

Additional information:
- The joints are all also driven around one axis and they get their drive target set before the physx step (so after deserialization even before the first step)
- I'm using the latest 4.1 branch (4.1.1.27006925)
- The `PxPhysics` and `PxScene` objects get deleted and recreated before deserialization, but I keep the `PxFoundation` and `PxDefaultCPUDispatcher` throughout the whole game.

Little tip for people who want to have multiple reusable restore points like we have: when you deserialize from a memory buffer physx will use this buffer while simulating and change it. So when you want to reuse the same restore point again it won't work. I tripped over that a little, but the solution is simple: copy the memory buffer, use the copy for deserialization and delete it when the simulation is finished.

kstorey-nvidia, 2021-01-26:
This is a known limitation of the reduced coordinate articulations at the moment. The joint state is stored in a reduced coordinate state that is not serialized by the automated collection system. It will be resolved in the future but it actually requires non-trivial changes to data layouts to support this.

As a workaround, you could extract the joint position/joint velocity from the articulation into a PxArticulationCache, save that state somewhere, then apply those cached joint position/velocity values to the articulation after it has been deserialized, restoring it to its original state. I realize that this is not an automatic process. Note that PxArticulationCache cannot be created or applied until the articulation has been inserted into a scene.

mrTag, 2021-01-26:
Thank you very much for the quick answer! That is good to know, should be no real problem to use the PxArticulationCache, only a minor inconvenience :)

One quick follow up question: can I use the PxArticulationCache that I create in one PxPhysics/PxScene context in the new PxPhysics and PxScene objects that I create for the deserialization? Or is there some connection in the PxArticulationCache to the actual memory of the scene? (I see some pointers in the PxArticulationCache class, but they are probably only for internal memory management?)

The articulation setup itself should be identical, I checked a few of the values I set and they all were deserialized correctly.

kstorey-nvidia, 2021-01-26:
Technically, you are only supposed to use a cache made from that particular articulation and there will be some checks that validate this in debug and checked build configurations, but if you are certain they will be compatible you should be able to work around these restrictions if you really want to. Ultimately, the cache just stores a sequence of numbers that correspond to joint positions and joint velocities.

mrTag, 2021-01-26:
That sounds like it should work in our situation. I'll just try it :)

Thanks again, very much appreciated.

closed 2021-01-26 by mrTag