NVIDIAGameWorks/PhysX #385: What exactly decides whether PVD is available?

rockyyliang, 2021-01-26:
Using the profile build of PhysX, which has PVD, the PVD only works when I set the config in Visual Studio to debug but does not work when I set it to release. Both configs link the profile build. 

So my question is, what condition does PhysX check to enable PVD? Is it a CMake flag?

Hoikas, 2021-02-02:
> PVD integration is enabled in the debug, checked and profiling configurations of the SDK. In order to reduce memory footprint and code size, it is not enabled in the release configuration.

See also the documentation: https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/VisualDebugger.html

rockyyliang, 2021-02-04:
> See also the documentation: https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/VisualDebugger.html

Yes you are correct that profile, debug, and checked configs have PVD. However, the availability of PVD still depends on what config you choose in Visual Studio. If you choose release, you don't get PVD even if you are linking one of these configs

I am **always** linking the profile config in CMake regardless of config selection:
```
set(physx_config "profile")
target_include_directories(target PUBLIC ${PHYSX_DIR}/physx/include)
target_include_directories(target PUBLIC ${PHYSX_DIR}/pxshared/include)
target_link_libraries(target ${PHYSX_DIR}/physx/bin/win.x86_64.vc141.md/${physx_config}/PhysX_64.lib)
target_link_libraries(target ${PHYSX_DIR}/PhysXCooking_64.lib)

```

Still trying to figure out the condition it uses to decide whether PVD is exposed

AlesBorovicka, 2021-02-05:
Hmm that does not sound correct, as a matter of fact I am working on a project that does have only debug/release configuration. In release we link against profile libraries as you do and PVD works. PVD enabled/disabled is based on compilation defines, defined in cmake (source/compiler/cmake/windows/CMakeLists.txt):

`SET(PHYSX_WINDOWS_DEBUG_COMPILE_DEFS   "PX_DEBUG=1;PX_CHECKED=1;${NVTX_FLAG};PX_SUPPORT_PVD=1" CACHE INTERNAL "Debug PhysX preprocessor definitions")
SET(PHYSX_WINDOWS_CHECKED_COMPILE_DEFS "PX_CHECKED=1;${NVTX_FLAG};PX_SUPPORT_PVD=1" CACHE INTERNAL "Checked PhysX preprocessor definitions")
SET(PHYSX_WINDOWS_PROFILE_COMPILE_DEFS "PX_PROFILE=1;${NVTX_FLAG};PX_SUPPORT_PVD=1" CACHE INTERNAL "Profile PhysX preprocessor definitions")
SET(PHYSX_WINDOWS_RELEASE_COMPILE_DEFS "PX_SUPPORT_PVD=0" CACHE INTERNAL "Release PhysX preprocessor definitions")`

You can try to step in to the connect. But if you link against profile, you should be able to connect to PVD.