NVIDIAGameWorks/PhysX #386: Generating Android Projects

GregSlazinski, 2021-01-27:
```
Preset parameter required, available presets:
(0) android-arm64-v8a <--- Android-19, arm64-v8a PhysX SDK
(1) android <--- Android-19, armeabi-v7a with NEON PhysX SDK
(2) vc12win32 <--- VC12 Win32 PhysX general settings
(3) vc12win64 <--- VC12 Win64 PhysX general settings
(4) vc14win32 <--- VC14 Win32 PhysX general settings
(5) vc14win64 <--- VC14 Win64 PhysX general settings
(6) vc15uwp32 <--- VC15 UWP 32bit PhysX general settings
(7) vc15uwp64 <--- VC15 UWP 64bit PhysX general settings
(8) vc15uwparm32 <--- VC15 UWP 32bit PhysX general settings
(9) vc15uwparm64 <--- VC15 UWP ARM 64bit PhysX general settings
(10) vc15win32 <--- VC15 Win32 PhysX general settings
(11) vc15win64 <--- VC15 Win64 PhysX general settings
(12) vc16win32 <--- VC16 Win32 PhysX general settings
(13) vc16win64 <--- VC16 Win64 PhysX general settings
Enter preset number: 1
Running generate_projects.bat android

C:\Esenthel\ThirdPartyLibs\PhysX\physx>exit /b 0
Using preset xml: buildtools/presets/public/android.xml
Target platform: android using compiler: clang
PM_CMakeModules_PATH: C:/Esenthel/ThirdPartyLibs/PhysX/physx/../externals/CMakeModules
PM_PATHS: C:/Esenthel/ThirdPartyLibs/PhysX/physx/../externals/CMakeModules;C:/Esenthel/ThirdPartyLibs/PhysX/physx/../externals/targa
Cmake: cmake.exe
Please provide path to android NDK in variable PM_AndroidNDK_PATH.
```

Getting this error:
`Please provide path to android NDK in variable PM_AndroidNDK_PATH.`

Where/how to set it?
I've edited this file:
https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/buildtools/presets/public/android.xml
by adding
`    <cmakeParam name="PM_ANDROIDNDK_PATH" value="C:\Progs\AndroidNDK" />`
and it didn't help

AlesBorovicka, 2021-01-27:
Hi,
 you should set that env variable or you can modify the generate_project.bat and set it there.
Regards,
Ales

GregSlazinski, 2021-01-27:
I was able to set the variable by editing the BAT file.
But now I'm facing different problem:
```
PS C:\Esenthel\ThirdPartyLibs\PhysX\physx> .\generate_projects.bat

C:\Esenthel\ThirdPartyLibs\PhysX\physx>exit /b 0
Preset parameter required, available presets:
(0) android-arm64-v8a <--- Android-19, arm64-v8a PhysX SDK
(1) android <--- Android-19, armeabi-v7a with NEON PhysX SDK
(2) vc12win32 <--- VC12 Win32 PhysX general settings
(3) vc12win64 <--- VC12 Win64 PhysX general settings
(4) vc14win32 <--- VC14 Win32 PhysX general settings
(5) vc14win64 <--- VC14 Win64 PhysX general settings
(6) vc15uwp32 <--- VC15 UWP 32bit PhysX general settings
(7) vc15uwp64 <--- VC15 UWP 64bit PhysX general settings
(8) vc15uwparm32 <--- VC15 UWP 32bit PhysX general settings
(9) vc15uwparm64 <--- VC15 UWP ARM 64bit PhysX general settings
(10) vc15win32 <--- VC15 Win32 PhysX general settings
(11) vc15win64 <--- VC15 Win64 PhysX general settings
(12) vc16win32 <--- VC16 Win32 PhysX general settings
(13) vc16win64 <--- VC16 Win64 PhysX general settings
Enter preset number: 1
Running generate_projects.bat android

C:\Esenthel\ThirdPartyLibs\PhysX\physx>exit /b 0
Using preset xml: buildtools/presets/public/android.xml
Target platform: android using compiler: clang
PM_CMakeModules_PATH: C:/Esenthel/ThirdPartyLibs/PhysX/physx/../externals/CMakeModules
PM_PATHS: C:/Esenthel/ThirdPartyLibs/PhysX/physx/../externals/CMakeModules;C:/Esenthel/ThirdPartyLibs/PhysX/physx/../externals/targa
Cmake: cmake.exe
Not searching for unused variables given on the command line.
CMake Deprecation Warning at C:/Esenthel/ThirdPartyLibs/PhysX/externals/cmakemodules/android/android.toolchain.cmake:187 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.19/Modules/CMakeDetermineSystem.cmake:104 (include)
  CMakeLists.txt:34 (project)

CMake Warning at C:/Esenthel/ThirdPartyLibs/PhysX/externals/cmakemodules/android/android.toolchain.cmake:323 (message):
  Could not determine machine name for compiler from
  C:/Progs/AndroidNDK/toolchains/aarch64-linux-android-4.9/prebuilt/windows-x86_64
Call Stack (most recent call first):
  C:/Esenthel/ThirdPartyLibs/PhysX/externals/cmakemodules/android/android.toolchain.cmake:550 (__DETECT_TOOLCHAIN_MACHINE_NAME)
  C:/Esenthel/ThirdPartyLibs/PhysX/externals/cmakemodules/android/android.toolchain.cmake:607 (__GLOB_NDK_TOOLCHAINS)
  C:/Program Files/CMake/share/cmake-3.19/Modules/CMakeDetermineSystem.cmake:104 (include)
  CMakeLists.txt:34 (project)

CMake Warning at C:/Esenthel/ThirdPartyLibs/PhysX/externals/cmakemodules/android/android.toolchain.cmake:323 (message):
  Could not determine machine name for compiler from
  C:/Progs/AndroidNDK/toolchains/arm-linux-androideabi-4.9/prebuilt/windows-x86_64
Call Stack (most recent call first):
  C:/Esenthel/ThirdPartyLibs/PhysX/externals/cmakemodules/android/android.toolchain.cmake:550 (__DETECT_TOOLCHAIN_MACHINE_NAME)
  C:/Esenthel/ThirdPartyLibs/PhysX/externals/cmakemodules/android/android.toolchain.cmake:607 (__GLOB_NDK_TOOLCHAINS)
  C:/Program Files/CMake/share/cmake-3.19/Modules/CMakeDetermineSystem.cmake:104 (include)
  CMakeLists.txt:34 (project)

CMake Warning at C:/Esenthel/ThirdPartyLibs/PhysX/externals/cmakemodules/android/android.toolchain.cmake:323 (message):
  Could not determine machine name for compiler from
  C:/Progs/AndroidNDK/toolchains/x86-4.9/prebuilt/windows-x86_64
Call Stack (most recent call first):
  C:/Esenthel/ThirdPartyLibs/PhysX/externals/cmakemodules/android/android.toolchain.cmake:550 (__DETECT_TOOLCHAIN_MACHINE_NAME)
  C:/Esenthel/ThirdPartyLibs/PhysX/externals/cmakemodules/android/android.toolchain.cmake:607 (__GLOB_NDK_TOOLCHAINS)
  C:/Program Files/CMake/share/cmake-3.19/Modules/CMakeDetermineSystem.cmake:104 (include)
  CMakeLists.txt:34 (project)

CMake Warning at C:/Esenthel/ThirdPartyLibs/PhysX/externals/cmakemodules/android/android.toolchain.cmake:323 (message):
  Could not determine machine name for compiler from
[... 167 lines cut ...]
  CMakeLists.txt:34 (project)

CMake Warning at C:/Esenthel/ThirdPartyLibs/PhysX/externals/cmakemodules/android/android.toolchain.cmake:323 (message):
  Could not determine machine name for compiler from
  C:/Progs/AndroidNDK/toolchains/x86_64-4.9/prebuilt/windows-x86_64
Call Stack (most recent call first):
  C:/Esenthel/ThirdPartyLibs/PhysX/externals/cmakemodules/android/android.toolchain.cmake:550 (__DETECT_TOOLCHAIN_MACHINE_NAME)
  C:/Esenthel/ThirdPartyLibs/PhysX/externals/cmakemodules/android/android.toolchain.cmake:607 (__GLOB_NDK_TOOLCHAINS)
  C:/Program Files/CMake/share/cmake-3.19/Modules/CMakeDetermineSystem.cmake:104 (include)
  CMakeLists.txt:34 (project)

CMake Error at C:/Esenthel/ThirdPartyLibs/PhysX/externals/cmakemodules/android/android.toolchain.cmake:616 (message):
  Could not find any working toolchain in the NDK.  Probably your Android NDK
  is broken.
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.19/Modules/CMakeDetermineSystem.cmake:104 (include)
  CMakeLists.txt:34 (project)

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
PS C:\Esenthel\ThirdPartyLibs\PhysX\physx>
```
I have cmake 3.19.1
Android NDK r19c

ayoub-belarbi, 2021-01-27:
Can you please try with any of the following NDK version equal or older than r11c (e.g. r11c -r11b -r11a -r10e -r10d -r10c -r10b -r10...)

Regards,
Ayoub

GregSlazinski, 2021-01-27:
I've updated cmake to latest version 3.19.3
Android NDK android-ndk-r22
Same problem.

Testing android-ndk-r11c, appears to work so far (generating the projects at least), but I'll have to test if it will compile, link and run.
Can you please add support for latest NDK? I'm concerned outdated NDK has some bugs/security issues.

ayoub-belarbi, 2021-01-27:
We definitely want to add native cmake support for Android but can't say exactly when to be honest. Will try to work on this in the upcoming weeks.

GregSlazinski, 2021-01-30:
Please look here, this would solve the problems:
https://github.com/NVIDIAGameWorks/PhysX/issues/390

CogSimGuy, 2021-06-04:
Any updates on this effort? I've spent days trying to work through building for Android v8a and nothing!