NVIDIAGameWorks/PhysX #387: Linux Compile Error

GregSlazinski, 2021-01-28:
```
In file included from /home/greg/Desktop/Esenthel/ThirdPartyLibs/PhysX/physx/source/pvd/src/PxProfileEventImpl.cpp:29:
In file included from /home/greg/Desktop/Esenthel/ThirdPartyLibs/PhysX/physx/source/pvd/src/PxProfileZoneImpl.h:38:
/home/greg/Desktop/Esenthel/ThirdPartyLibs/PhysX/physx/source/pvd/src/PxProfileEventBufferAtomic.h:293:5: error: use of function '__builtin_alloca' is discouraged; there is no way to check for failure but failure may still occur, resulting in a possibly exploitable security vulnerability [-Werror,-Walloca]
                                PX_ALLOCA(tempBuffer, uint8_t, sizeToWrite);
                                ^
/home/greg/Desktop/Esenthel/ThirdPartyLibs/PhysX/physx/source/foundation/include/PsAlloca.h:73:43: note: expanded from macro 'PX_ALLOCA'
                        var.mPointer = reinterpret_cast<type*>(PxAlloca(size));                                                    \
                                                               ^
/home/greg/Desktop/Esenthel/ThirdPartyLibs/PhysX/physx/source/foundation/include/PsAllocator.h:101:22: note: expanded from macro 'PxAlloca'
        #define PxAlloca(x) alloca(x)
                            ^
/usr/include/alloca.h:35:23: note: expanded from macro 'alloca'
# define alloca(size)   __builtin_alloca (size)
                        ^
1 error generated.

```
clang version 10.0.0-4ubuntu1 
Target: x86_64-pc-linux-gnu

Malicestriker, 2021-02-01:
I met the same problem, is there any solution? thx!

GregSlazinski, 2021-02-01:
I've integrated all known fixes and workarounds over to my game engine's version:
https://github.com/Esenthel/EsenthelEngine/tree/master/ThirdPartyLibs/PhysX

Hoikas, 2021-02-02:
PhysX compiles with `-Werror` by default. You can remove it with this patch or by using the physx port in vcpkg.

```diff
diff --git a/physx/source/compiler/cmake/linux/CMakeLists.txt b/physx/source/compiler/cmake/linux/CMakeLists.txt
index 13acf52e..b709b9f9 100644
--- a/physx/source/compiler/cmake/linux/CMakeLists.txt
+++ b/physx/source/compiler/cmake/linux/CMakeLists.txt
@@ -28,8 +28,8 @@
 STRING(TOLOWER "${CMAKE_BUILD_TYPE}" CMAKE_BUILD_TYPE_LOWERCASE)
 
 #TODO: Fix warnings
-SET(CLANG_WARNINGS "-ferror-limit=0 -Wall -Wextra -Werror -Wstrict-aliasing=2 -Weverything -Wno-documentation-deprecated-sync -Wno-documentation-unknown-command -Wno-gnu-anonymous-struct -Wno-undef -Wno-unused-function -Wno-nested-anon-types -Wno-float-equal -Wno-padded -Wno-weak-vtables -Wno-cast-align -Wno-conversion -Wno-missing-noreturn -Wno-missing-variable-declarations -Wno-shift-sign-overflow -Wno-covered-switch-default -Wno-exit-time-destructors -Wno-global-constructors -Wno-missing-prototypes -Wno-unreachable-code -Wno-unused-macros -Wno-unused-member-function -Wno-used-but-marked-unused -Wno-weak-template-vtables -Wno-deprecated -Wno-non-virtual-dtor -Wno-invalid-noreturn -Wno-return-type-c-linkage -Wno-reserved-id-macro -Wno-c++98-compat-pedantic -Wno-unused-local-typedef -Wno-old-style-cast -Wno-newline-eof -Wno-unused-private-field -Wno-format-nonliteral -Wno-implicit-fallthrough -Wno-undefined-reinterpret-cast -Wno-disabled-macro-expansion -Wno-zero-as-null-pointer-constant -Wno-shadow -Wno-unknown-warning-option -Wno-atomic-implicit-seq-cst -Wno-extra-semi-stmt")
-SET(GCC_WARNINGS "-Wall -Werror -Wno-invalid-offsetof -Wno-uninitialized")
+SET(CLANG_WARNINGS "-ferror-limit=0 -Wall -Wextra -Wstrict-aliasing=2 -Weverything -Wno-documentation-deprecated-sync -Wno-documentation-unknown-command -Wno-gnu-anonymous-struct -Wno-undef -Wno-unused-function -Wno-nested-anon-types -Wno-float-equal -Wno-padded -Wno-weak-vtables -Wno-cast-align -Wno-conversion -Wno-missing-noreturn -Wno-missing-variable-declarations -Wno-shift-sign-overflow -Wno-covered-switch-default -Wno-exit-time-destructors -Wno-global-constructors -Wno-missing-prototypes -Wno-unreachable-code -Wno-unused-macros -Wno-unused-member-function -Wno-used-but-marked-unused -Wno-weak-template-vtables -Wno-deprecated -Wno-non-virtual-dtor -Wno-invalid-noreturn -Wno-return-type-c-linkage -Wno-reserved-id-macro -Wno-c++98-compat-pedantic -Wno-unused-local-typedef -Wno-old-style-cast -Wno-newline-eof -Wno-unused-private-field -Wno-format-nonliteral -Wno-implicit-fallthrough -Wno-undefined-reinterpret-cast -Wno-disabled-macro-expansion -Wno-zero-as-null-pointer-constant -Wno-shadow -Wno-unknown-warning-option -Wno-atomic-implicit-seq-cst -Wno-extra-semi-stmt")
+SET(GCC_WARNINGS "-Wall -Wno-invalid-offsetof -Wno-uninitialized")
 
 IF ("${CMAKE_CXX_COMPILER_ID}" STREQUAL "Clang")
 	# using Clang	  
```

Malicestriker, 2021-02-04:
@Hoikas thanks a lot, it works well!

ayoub-belarbi, 2021-04-22:
Thank you for reporting this issue! We just released PhysX 4.1.2 that fixes this issue (along with other fixes). Make sure to check it out :) 
Your contribution is greatly appreciated!

closed 2021-04-22 by ayoub-belarbi