NVIDIAGameWorks/PhysX #394: Including Coriolis and centrifugal forces with PxRigidDynamic

chrisdembia, 2021-02-11:
The paper [Simulation Tools for Model-Based Robotics: Comparison of Bullet, Havok, MuJoCo, ODE and PhysX](https://homes.cs.washington.edu/~todorov/papers/ErezICRA15.pdf) from 2015 mentions that PhysX omits Coriolis forces, based on a [PhysX forum post](https://forums.developer.nvidia.com/t/are-eulers-equations-integrated-for-a-pxrigiddynamic-in-pxscene-simulate-dt/27677) from 2012.

I have a few questions:

1. Is it still true that PhysX omits Coriolis forces? The PhysX documentation indicates that Coriolis (and centrifugal) forces are included only with reduced coordinate articulations.
2. Has anyone tried editing the PhysX source code to include Coriolis and centrifugal forces for `PxRigidDynamic`?
3. Is it possible to use a rigid body that includes Coriolis and centrifugal forces with the Vehicle SDK? It seems that the Vehicle SDK requires a `PxRigidDynamic`, and could not accept a `PxArticulationLink`.

kstorey-nvidia, 2021-02-11:
It is true that in PhysX 4, only PxArticulationLink includes Coriolis and Centrifugal forces. In PhysX 5, these forces will be included via an optional rigid body flag so that legacy behavior can be maintained if required.

What is the specific vehicle you need to simulate that you feel Coriolis/Centrifugal forces would be required? For most vehicles, these effects would be minor.

chrisdembia, 2021-02-12:
Thank you for the quick response! I'm interested in simulating tractor-trailer vehicles.