NVIDIAGameWorks/PhysX #399: PxVehicleSetMaxHitActorAcceleration applied on vehicle?

zemlifr, 2021-03-10:
Hi, we had problem with very small objects flying far away when hit by vehicle wheels. We resolved it by setting

`PxVehicleSetMaxHitActorAcceleration` to some small value, but now it seems when wheel is on some dynamic object, it goes visually through car chassis. Like if suspension force is not large enough tu move vehicle up and only wheel shape is repositioned.

I have looked into PxVehicleUpdate code and I found a part where limit is applied
......
```
					//Compute the total force acting on the suspension.
					//Remember that the spring force acts along -w.
					//Remember to account for the term perpendicular to w and that it acts along -TTimesBeta
					PxF32 suspensionForceMag = hitNorm.dot(-w*suspensionForceW - TTimesBeta*susp.mSprungMass);

					//Apply the opposite force to the hit object.
					//Clamp suspensionForceMag if required.
					if (dynamicHitActor && !(dynamicHitActor->getRigidBodyFlags() & PxRigidBodyFlag::eKINEMATIC))
					{
						const PxF32 dynamicActorInvMass = dynamicHitActor->getInvMass();
						const PxF32 dynamicActorMass = dynamicHitActor->getMass();
						const PxF32 forceSign = computeSign(suspensionForceMag);
						const PxF32 forceMag = PxAbs(suspensionForceMag);
						const PxF32 clampedAccelMag = PxMin(forceMag*dynamicActorInvMass, gMaxHitActorAcceleration);
						const PxF32 clampedForceMag = clampedAccelMag*dynamicActorMass*forceSign;
						PX_ASSERT(clampedForceMag*suspensionForceMag >= 0.0f);

--------THERE-->			suspensionForceMag = clampedForceMag;

						hitActors[i] = dynamicHitActor;
						hitActorForces[i] = hitNorm*(-clampedForceMag*timeFraction);
						hitActorForcePositions[i] = hitContactPoints4[i];
					}

.....
```

It seems weird to me, that clampedForce is applied not only on hitActor, but also on vehicle itself.
From comment in `PxVehicleSetMaxHitActorAcceleration` i would expect otherwise.

Or do I understand it wrong?

gyeomannvidia, 2021-03-11:
I agree that the doxy is rather ambiguous on this point.

The reason that the force is applied to both is to preserve Newton's laws. To be honest, this is a bit fussy because limiting the acceleration already breaks them :)   I think this feature should be extended to decide how the force is applied to each object.  You've raised a good point.

Can you try leaving   suspensionForceMag  unchanged  and see if it resolves your issue?

zemlifr, 2024-04-12:
Yes, that works for us

closed 2024-04-12 by zemlifr