NVIDIAGameWorks/PhysX #401: how to import TriangleMesh into gScene and can be visable in windows?

Timkeeper2018, 2021-03-14:
I tried to introduce the "createBV34TriangleMesh" function in the "SnippetTriangleMeshCreate" example into SnippetHelloWorld, to create a random terrain data. I added the following statement at the end of the function:

`PxTriangleMeshGeometry TriangleMeshGeo;
TriangleMeshGeo.triangleMesh = triMesh;
PxShape* shape = gPhysics->createShape(TriangleMeshGeo, *gMaterial);

PxTransform localTm(0.0, 1.0, 0.0);
PxRigidDynamic* body = gPhysics->createRigidDynamic(localTm);
body->attachShape(*shape);

gScene->addActor(*body);

shape->release();`

But I cannot see the random terrain in the window.
But when I loaded ConvexMesh, there was no such problem. Through the above code, ConvexMesh can display it in the window well.
What should I do to make TriangleMesh appear in the window with physical effects? Thanks in advance.

Timkeeper2018, 2021-03-14:
当我使用如下代码的时候，创建一个静态的刚体

		PxTriangleMesh* mesh = createMeshGround();
		PxTriangleMeshGeometry geom(mesh);
		PxRigidStatic* groundMesh = gPhysics->createRigidStatic(PxTransform(PxVec3(0, 1.0, 0)));
		PxRigidActorExt::createExclusiveShape(*groundMesh, geom, *gMaterial);
		gScene->addActor(*groundMesh);

而当我把PxRigidStatic修改为PxRigidDynamic的时候动态刚体没法与地面进行碰撞检测，直接穿过地面，下沉下去。
地面是通过如下方法建立的：

PxRigidStatic* groundPlane = PxCreatePlane(*gPhysics, PxPlane(0,1,0,0), *gMaterial);
	gScene->addActor(*groundPlane);

Timkeeper2018, 2021-03-14:
When I use the following code, create a static rigid body

PxTriangleMesh* mesh = createMeshGround();
PxTriangleMeshGeometry geom(mesh);
PxRigidStatic* groundMesh = gPhysics->createRigidStatic(PxTransform(PxVec3(0, 1.0, 0)));
PxRigidActorExt::createExclusiveShape(*groundMesh, geom, *gMaterial);
gScene->addActor(*groundMesh);

When I changed PxRigidStatic to PxRigidDynamic, the dynamic rigid body could not collide with the ground, and passed directly through the ground and then sank.
The ground is established by the following methods:

PxRigidStatic* groundPlane = PxCreatePlane(*gPhysics, PxPlane(0,1,0,0), *gMaterial);
gScene->addActor(*groundPlane);

kstorey-nvidia, 2021-03-15:
PhysX does not support non-convex triangle meshes as a collision primitive on dynamic actors. TriMesh, plane and heightfield can only be used as collision primitives for static/kinematic actors or as scene query primitives for all rigid actor types. The reason for this is that PhysX does not provide collision methods for these primitives. This limitation is covered in the guide.

Timkeeper2018, 2021-03-15:
Thanks a lot

closed 2021-03-15 by Timkeeper2018