NVIDIAGameWorks/PhysX #403: Kinematic CCT and dynamic rigid actors

Esentiel, 2021-03-17:
Hello!

Have a question regarding implementation of slope limit for dynamic rigid actor (most likely kinematic).
I read that slope limit only works for static actors as of now. But in real world, there are plenty case when we need to have slope limit against dynamic kinematic actors. How can that be done?
I made some naive implementation:
Inside onShapeHit() I rast several rays in -upVector direction against touched dynamic kinematic actor, build several triangles within hit points and calculate mean normal among those triangles normals. 
This works in most case. Beside that it's reactional slope limit: I allow player to move onto non-Walkable plane and then force slide him. You can imagine, this produces trembling if I face against slope and hold move-forward-button. On the other hand, slope limit for static actors is implemented in the way it just won't let to step on non walkable tri, in this case I got no trembling.

On the one hand, I 'm avoiding physx code changing really hard (except spot little fixes) and don't want to extend/modify cct code very much. On the other hand, I want to have single behavior for static and dynamic actors.
I thought I can decline current result of move() function if it produces slide effect (move player back to his prev position) but not sure If I can cover all the cases ( jump onto slope, move in the slope direction from the side, or from the bottom or from the top position).

PS: I read about user-defined obstacle that can be generated in runtime for cct, but I couldn't find any usage for it, maybe I messed something and I can utilize this technique to reach my goal.

Appreciate any help or advice!

Esentiel, 2021-03-17:
Actually I just realized that slope limit implementation for static doesn't look to good on our game designers look, unfortunately.
Player stuck (literally can't move) if he touches non-walkable tris from slope that are on the side of the player. I expected it will just cut movement of the direction that leads to non-walkable tris and let player move on tangent to the slope.
I draw this to explain the situation, pls forgive my paint skills

[image]

I assume, major code changes are required to modify slope limit functionality to make it behave like we want in our game.

PierreTerdiman, 2021-03-18:
Let's address the problem against static objects first. It is not normal that the character gets stuck. In SampleBridges you can change this line to enable the slope limit against static objects:

https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/samples/samplebridges/SampleBridges.cpp#L59

Then move to that piece of geometry (close to the starting point) in the attached screenshot. The character should slide along that non-walkable slope, without getting stuck.

[image]

Esentiel, 2021-03-18:
Yes, that works for me in Sample. Will debug to find the difference, probably some other code path was selected in my case.

Thanks for help!

Esentiel, 2021-03-18:
So to reproduce my problem, pls uncomment the following line:
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/samples/samplecctsharedcode/SampleCCTActor.cpp#L166

I didn't find the place in code that produce such behavior, yet, but now you should be able to reproduce the issue.

PierreTerdiman, 2021-03-18:
**_does physx utilize spatial world/tree/pattern at any rate to optimize scene queries like raycasts, sweeps, overlaps?_**

Of course.

**_I heard that's the case for static but not for dynamic actors, can you confirm this pls?_**

This is incorrect (and it would be strange). PhysX uses spatial structures for both. In fact you can explicitly choose which ones are used for static & dynamic objects in PxSceneDesc when creating the scene.

**_and maybe you can point me into code that implements this(wasn't able to find it),_**

Generally speaking the scene query code is here: https://github.com/NVIDIAGameWorks/PhysX/tree/4.1/physx/source/scenequery

But this is just the higher-level scene structures. It also uses a lot of low-level intersection / overlap / sweep functions located here: https://github.com/NVIDIAGameWorks/PhysX/tree/4.1/physx/source/geomutils/src

**_from what I've seen it performs overlap to generate cache and to find list of possible touched shapes then tests every of them using sweep._**

Indeed. There is a scene-level overlap query to fill caches, and then it sweeps against the caches. The caches are usually not re-filled every frame, especially the static cache.

**_And I was thinking about using such technique for cct overlap query._**

It would certainly be possible to customize the code to take advantage of whatever spatial structures you already have in place. Do some profiling before though, to make sure the overlap/sweeps are actually the source of any performance problem.

Esentiel, 2021-03-19:
So, I found cases that don't let me use this approach with several raycasts for dynamic actors to get the angle of the plane that are beneath cct.

If I got no any hits back(actually less than 3, so I can't build triangle) from raycast I assumed that this is the angle of plane close to pi/2 and forced slide using PxControllerBehaviorFlag::eCCT_SLIDE

This resolved 1st case from below image. But raised the second - I can't step on dynamic actors because it forces slide due to same input data: 2 hits only from raycasts. If I roll back this fix, cct will be able to climb up on 85 degrees slopes.
[image]

Back to basics, As far as I understood, slope limit was implemented n PhysX only for static actors because it's very easy to find the angle having touched triangle. Dynamic actors don't use trimesh (at least in our engine), but it uses primitives (sphere, capsule, box) and convexmesh. I assume it should be possible to implement algorithm close to that we already have for trimesh - find touched convex, get it's normal -> calculate angle and height (for step functionality). For primitives it should not be a problem either, I guess.

Before i dived into researching this, maybe you can give couple hints why this wasn't implemented? I'm afraid there might be some pitfalls, probably performance issues? Maybe using your experience with PhysX you would be able to advise another options rather than copying trimesh algorithm?

thanks!

Esentiel, 2021-03-19:
Done with the commit https://github.com/Esentiel/PhysX/commit/d1e38c5493682a8c1e742a56a5eff35fc6286441
Now all I need is to write custom logic for handling primitives.
I guess it should be relatively easy knowing primitive params, I should be able understand the plane angle. with sphere and capsule it should be normal to tangent at hit point probably, for box I will find the solution next week, probably some calculations based on vertices and normals against hit point in ohShapeHit() info

closed 2021-03-23 by Esentiel