NVIDIAGameWorks/PhysX #404: Persistent contact reports zero linear velocity change

pkusilence, 2021-03-17:
I am testing Extra Contact Data with PhysX 3.4.2 included in UE4.26.1 by raising flag ePOST_SOLVER_VELOCITY and ePRE_SOLVER_VELOCITY.  

A sphere with restitution = 0 and friction = 0,  is set Vx=200cm/s in the Tick function (which is ticking while PhysX is not simulating) .  Then the sphere collides with a plane. The result is :  when the touch is found at first, the pre Vx=200 cm/s and the post Vx=16cm/s (which seems ok for me);  when the contact keeps ( persistent) , the pre and post velocities remain exact the same. Which maybe means the solver does not change the velocity. However, the Vx is getting close to zero as the simulation step forward.

So I followed engine works , finding:  for PersistentContact case,  the contact solver is excute after velocity is written into the stream ; for touch found, the solver is before write.

Could any developer can explain these phenomena?

kstorey-nvidia, 2021-03-18:
I will file an internal bug to investigate because this seems like a bug.

One thing to note is that contact reports are reported before the velocity state has been copied to the buffered/external state of the actors, so you can actually get the pre-contact velocity from the respective actors by calling getLinearVelocity()/getAngularVelocity(). The post-contact velocity reported should be correct, so you should be able to work around this to get the expected results.

pkusilence, 2021-03-18:
Thanks for your reply, it does works, just need to handle vz -= gravity*dt.

TwentyPast4, 2021-03-25:
I'm also experiencing this problem. Post and Pre seem to  indicate a zero velocity change, while the normal impulse is non-zero

closed 2024-01-25 by pkusilence