NVIDIAGameWorks/PhysX #406: Very misleading image in guide for sweep

MaxLap, 2021-03-21:
Hello,

At https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/GeometryQueries.html, there is this image explaining what a sweep is:

[image]

The "hit distance" is wrong. It should be from the middle of the right shape to the middle of its "shadow" on the left. 

I had been using unnecessary hacks, such as doing a reverse RayCast, to figure out which part of the shape touched, because this image mislead me. At one point, I just though that this hit distance was so useless it had to be a mistake in the image. And after testing, the mistake is indeed in the image.

I'm sure I'm not alone in this situation. Could someone fix the image please? Thank you

Link to the image in this repo: https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/documentation/PhysXGuide/_images/GeomQuerySweep.png

alanjfs, 2021-03-27:
Did some quick surgery to better understand what it was *supposed* to look like, is this what you mean?

[image]

MaxLap, 2021-03-27:
Yes, this looks perfect!

PierreTerdiman, 2021-03-29:
Thank you, this image is indeed incorrect. Will fix.

kstorey-nvidia, 2021-04-22:
This should be corrected now

closed 2021-04-22 by kstorey-nvidia