NVIDIAGameWorks/PhysX #410: Where is PxParticleFluid* in 4.x?

f1nalspace, 2021-04-11:
I want to port over my "Fluid Sandbox" application shown here (https://www.youtube.com/watch?v=eUbwxTE38TY) from PhysX 3.3.3 to the latest PhysX SDK (4.1) from this GitHub repository. But looking at the include files, there is no "PxParticleFluid" anymore. As a matter of fact, searching for "particle" does not show any includes for particles at all. Are particles removed from the 4.x branch?

f1nalspace, 2021-04-11:
Never mind, found the migration guide in the 3.4 branch and it looks like you have dropped the particles support entirely - sad.
Maybe I should port it over to Flex or write my own physics system...

https://github.com/NVIDIAGameWorks/PhysX-3.4/blob/master/PhysX_3.4/Documentation/PhysXGuide/_sources/Manual/MigrationTo34.txt

kstorey-nvidia, 2021-04-12:
Particle and cloth support was removed in PhysX 4. Neither the cloth or particle implementations had full interactivity with the rigid body scene, so they didn't offer much in terms of functionality that couldn't be achieved by using either NvCloth or Flex, which is what we would recommend you use for now.

Particles and cloth will be back in PhysX 5 as part of a new unified solver.

f1nalspace, 2021-04-12:
> Particle and cloth support was removed in PhysX 4. Neither the cloth or particle implementations had full interactivity with the rigid body scene, so they didn't offer much in terms of functionality [...]

Okay for now I will use PhysX 3.4 - it still has PxParticleFluid and works just fine ;)
But I want to give Flex a try. Are there any guides for migrating from PhysX to Flex?

Also i am looking forward to PhysX 5.

Zeblote, 2021-04-12:
> Particles and cloth will be back in PhysX 5 as part of a new unified solver.

But when, though? Are we going to be waiting another year with zero news?

Hoikas, 2021-04-13:
> But when, though? Are we going to be waiting another year with zero news?

Obviously yes. The silence is deafening.