NVIDIAGameWorks/PhysX #412: Math classes

fido9dido, 2021-04-25:
The documentation mentioned math classes in foundation library such as PxVec4, PxMat44 
[I](https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/API.html)
but they are not implemented with intrinsics, where can I find the ones implemented with intrinsics????

kstorey-nvidia, 2021-04-26:
The PxVec4 etc. maths types do not use intrinsics and are mainly used as interface types.

Much of the low-level dynamics and collision code uses intrinsics where they can offer performance advantages. The underlying intrinsics are platform-specific. For example, the Windows-specific files are found here in the foundation code here:

https://github.com/NVIDIAGameWorks/PhysX/tree/4.1/physx/source/foundation/include/windows
https://github.com/NVIDIAGameWorks/PhysX/tree/4.1/physx/source/foundation/src/windows

chris-chan-commits, 2021-04-26:
use glm

closed 2021-04-27 by fido9dido