NVIDIAGameWorks/PhysX #414: Porting PhysX to another platform

Esentiel, 2021-04-28:
Hello!

PhysX doesn't support PS4(or BSD which should be close to PS4) and probably we will need to port it to PS4, can anyone involved in the project(or the one who was involved in the similar task) please estimate how much time will it take with any possible accuracy(days/weeks). I checked some of platform specific stuff that I faced to trying to compile sdk on BSD, and that didn't look to scary. On the other hand, I expect some pitfalls that I can't see now.

Any feedback will be very helpful!

Esentiel, 2021-04-28:
I found similar issue and answer in it.

kstorey-nvidia, 2021-04-30:
If you send an email to physxlicensing@nvidia.com and can confirm that you are a signed up console developer, you should be able to get access to the platform-specific code and cmake configurations for consoles. Due to vendor NDAs etc., we can't provide those files freely on GitHub.

closed 2021-05-17 by Esentiel