NVIDIAGameWorks/PhysX #415: Compile error with latest Xcode

Ravbug, 2021-04-30:
When compiling PhysX on Xcode 12.5, compile fails with:

```cpp
In file included from /Users/ravbug/Documents/RavEngine-Samples/RavEngine/deps/physx/physx/source/geomutils/src/mesh/GuRTreeQueries.cpp:53:
In file included from /Users/ravbug/Documents/RavEngine-Samples/RavEngine/deps/physx/physx/source/geomutils/src/mesh/GuRTree.h:45:
In file included from /Users/ravbug/Documents/RavEngine-Samples/RavEngine/deps/physx/physx/source/foundation/include/PsVecMath.h:1338:
In file included from /Users/ravbug/Documents/RavEngine-Samples/RavEngine/deps/physx/physx/source/foundation/include/PsInlineAoS.h:40:
In file included from /Users/ravbug/Documents/RavEngine-Samples/RavEngine/deps/physx/physx/source/foundation/include/unix/PsUnixInlineAoS.h:40:
/Users/ravbug/Documents/RavEngine-Samples/RavEngine/deps/physx/physx/source/foundation/include/unix/neon/PsUnixNeonInlineAoS.h:3582:10: error: argument value 2 is outside the valid range [0, 1]
                return vdupq_lane_f32(vget_low_f32(a), index);
```

The line of code that originates the error is 106 in `GuRTreeQueries.cpp`
```cpp
Vec4V nqMinx4 = V4SplatElement<0>(nqMin);
Vec4V nqMiny4 = V4SplatElement<1>(nqMin);
Vec4V nqMinz4 = V4SplatElement<2>(nqMin);       //fails here
Vec4V nqMaxx4 = V4SplatElement<0>(nqMax);
Vec4V nqMaxy4 = V4SplatElement<1>(nqMax);
Vec4V nqMaxz4 = V4SplatElement<2>(nqMax);
```

Compilation works on earlier versions of Xcode and on latest Visual Studio.

DeanBilotti, 2021-05-23:
Looks like the PX_UWP platform probably had the same issue with the template definition for V4SplatElement so they split out the cases for 0 and 1. Taking the same PX_UWP path in PsUnixNeonInlineAoS.h works when using Xcode 12.5.

DeanBilotti, 2021-05-23:
Or constexpr would work in c++17: `if constexpr(index < 2)`

DeanBilotti, 2021-05-23:
For this function it's probably worth all compilers taking the PX_UWP path to be honest I don't think a c++ compiler is required to omit code before doing error checking like this.

DeanBilotti, 2021-05-24:
Yes it looks like clang was updated to detect incorrect intrinsic parameters but it isn't smart enough to prevent the diagnostic in this case. I assume that was the case for PX_UWP too so either use that path or fix with constexpr (c++17).