NVIDIAGameWorks/PhysX #420: PhysX 4.1.2 - Overly enthusiastic Box-Box contacts

alanjfs, 2021-05-07:
Hi all,

Since the latest update, I've been noticing boxes in particular can get real excited whenever there's contact.

- Semi-related: #409 

[image]

I suspect friction, but can't put my finger on why?

- Friction on boxes: 0.8
- Friction on ground: 0.5
- Restitution on boxes: 0.1
- Restitution on ground: 0.0

Any thoughts on where I should start looking if I'm happy with the amount of friction elsewhere? Maybe the restOffset or contactOffset are to blame?

I would have liked to capture a PVD of the above for you, but unfortunately the behavior *differs* in a Debug build. :S I did however manage to reproduce the behavior in less than 5 seconds of adjusting the initial state.

- [enthusiasticPhysX.zip](https://github.com/NVIDIAGameWorks/PhysX/files/6440001/enthusiasticPhysX.zip)

[image]

Any ideas?

Thanks!

kstorey-nvidia, 2021-05-07:
I don't see anything obviously wrong with your setup, so I think this issue is likely related to the error corrective components of the TGS solver. There will be a small amount of penetration, but, due to small dt and relatively large iteration counts, the velocity introduced to correct the penetration can be over-enthusiastic. This over-enthusiastic response has been improved a lot in PhysX 5, but it is too much of a significant solver change to consider back-porting to 4.1

A good option to resolve this would be to set "max depenetration velocity" on your rigid bodies to a smaller number than default. The default is infinite. This clamps the magnitude of the impulses used to correct penetrations. You don't want to set this too low as it will lead to spongy contacts. A good ballpark value for this would be around ~2m/s. Give it a try and let me know if it helps.

You could also try increasing contact offset or enabling speculative CCD, but I don't expect this will be the source of the issues because contacts did not appear to be being missed due to fast motion in this case.

alanjfs, 2021-05-07:
Bingo!

[image]

Past 2.0, I stopped noticing any difference. In fact, I was expecting the box to eventually stop responding to contacts at really low values like 0.0001, but not much else changed. Maybe depenetration isn't the only thing keeping it from falling through the floor? I also messed around with Max Contact Impulse at 0.0001 but that also didn't seem to have any effect.

That's besides this issue however, thanks for the swift reply, I'm happy now 💘