NVIDIAGameWorks/PhysX #422: Integration Scheme and LCP

HA1N, 2021-05-17:
Hello Devs,

I'm doing a comparison between Unity3D and Matlab to explore the capability of the PhysX as a robot simulation environment.
I want to give a comprehensive comparison between matlab and PhysX to show the advantages and disadvantages of both programs and hopefully show that physics engines nowadays can be used for accurate robot simulation with less computation time.

To accurately compare the results of PhysX I need to know the integration scheme used to solve the Equations of Motion (EOM) so I can set up matlab using the same method. I tried to find it in the source code but wasn't successful. Can you please show me where to find the used integration method?

I'm mainly working with the articulation-body feature. For that the joint constraints and EOM need to be formulated in a Linear Complementarity Problem (LCP) to be solved by the PGS or TGS solver. I haven't found the used approach in the source code but I assume it is similar to the Conastraint-Based Approach presented by K. Erleben: https://www.researchgate.net/profile/Kenny-Erleben/publication/247181209_Physics-Based_Animation/links/5e1b2ed04585159aa4cb43d8/Physics-Based-Animation.pdf

Thanks in advance!

kstorey-nvidia, 2021-05-17:
The integration method is backward Euler. There are two parts for articulations - computeUnconstrainedVelocities computes the accelerations acting on the articulation and integrates these into the velocities, constraints are solved by the chosen solver (PGS/TGS) on the velocity level, then positions are integrated using the final velocities produced by the solver. The underlying approach used is based on Featherstone's ABA algorithm.

Contacts, limits and drives are solved using a constraint-based approach. Contacts and hard limits use a hard constraint model. Joint drives and soft limits use an unconditionally stable implicit spring-damper model.

HA1N, 2021-05-17:
Thanks a lot, that is exaclty what i was missing!

HA1N, 2021-07-13:
Hello,

I have some new questions regarding the theoretical background of PhysX. 
- What exactly is meant by hard/soft limits?
- Is the unconditionally stable implictit spring-damper model a normal ordinary differential equation which is used by an implicit scheme i.e. Backward Euler?

Another look at Erlebens book "Physics-based Animation" and the constraint-based approach described there shows some differences to the approach explained in the last post. Here the following equations for the position and velocity are solved:
[image]

You explained that the unconstained velocities are computed by integrating the accelerations. And afterwards the constraints are solved on velocity level to get the "real" velocities which are used to get the new position of the articulated body. This should be identical to the approach by Erleben besides the computation of the unconstained velocities?