NVIDIAGameWorks/PhysX #425: UE5 PhysX fork

Zeblote, 2021-05-27:
With Epic somehow actually planning to deprecate PhysX in favor of the incredibly slow Chaos physics engine, many people are going to be in need of a usable physics solution for UE5.

Will nvidia maintain a branch of UE5 with PhysX going forward?

amoravanszky, 2021-06-14:
I have no plan to do so.  First, we are unfamiliar with the integration code as it was written by Epic's engineers.  Second, our past experience is that to keep any significant modification (e.g. our HairWorks comes to mind) up to date in UE is a rather Sisyphean effort given the frequency with which Epic makes significant changes to the underlying engine.  I would rather spend the same time to build new tech.  Third, if Epic had desired to support multiple physics engines in UE, they would have created an appropriate abstraction.  I don't see any indication that there is a desire to give users any choice in this matter, and I feel working against the grain here would just cause upset.

amoravanszky, 2021-06-14:
On the Unity front things are bit more promising, in part because their scripting level abstractions are much less volatile.  I have been working with handing over the sources of an experimental Unity integration we have done to the Unity community for maintenance, this should become public soon.

Zeblote, 2021-06-14:
That's a shame, seeing how PhysX is clearly far superior to chaos...

closed 2021-06-21 by amoravanszky

mortonsykes, 2025-03-18:
Its 2025 and PhysX is still 6x faster in the project I ported to UE5.