NVIDIAGameWorks/PhysX #428: physx::PxShape::setGeometry not working

TheDonsky, 2021-06-01:
Hello,
I tried to resize a box shape by calling setGeometry(), but neither the behavior, nor the debuggers (PVD and MSVS) shows any change. Do I need to have some specific configuration for it to work?
I'm attaching a screenshot from visual studio, highlighting the issue.

When I run the sample code below, 'old' always equals 'cur' and they both differ from 'desired'.
```
physx::PxBoxGeometry old;
bool gotOld = Shape()->getBoxGeometry(old);
physx::PxBoxGeometry desired = Geometry(newShape);
Shape()->setGeometry(desired);
physx::PxBoxGeometry cur;
bool gotCur = Shape()->getBoxGeometry(cur);
```

[image]

kstorey-nvidia, 2021-06-01:
Are you using shared shapes? It's illegal to change the geometry on a shared shape once it is attached to an actor. You should get some debug error messages telling you what you did that isn't allowed.

TheDonsky, 2021-06-01:
Thanks, I was not aware of the restriction and logs were not visible when I first stumbled across the issue for some reason and did not think of looking when I switched to the debug build..

closed 2021-06-01 by TheDonsky