NVIDIAGameWorks/PhysX #431: Linking error Physx 4.1

Bicyclette, 2021-06-11:
Hi everyone ! I am a new Physx user and I try to integrate it in some existing C++ code.
Currently I get the following error with the makefile generated from my CMake script :

> /usr/bin/ld: libPhysX_static_64.a(PxPhysXGpuModuleLoader.cpp.o): undefined reference to symbol 'dlsym@@GLIBC_2.2.5'
> /usr/bin/ld : /lib/x86_64-linux-gnu/libdl.so.2 : erreur lors de l'ajout de symboles : DSO missing from command line
> collect2: error: ld returned 1 exit status

I am using a Linux operating system, and I have read some threads of people having the same issue.
They fixed this by changing the linking order for some Physx libraries, but it didn't changed anything for me.

Here is a fragment of CMake script :

``` cmake
include_directories(${PROJECT_SOURCE_DIR}/include ${PROJECT_SOURCE_DIR}/include/physx)

link_directories(${PROJECT_SOURCE_DIR}/libs/physx/debug)

target_link_libraries(${PROJECT_NAME} PhysXExtensions_static_64 PhysX_static_64 PhysXVehicle_static_64 PhysXCharacterKinematic_static_64 PhysXCooking_static_64 PhysXCommon_static_64 PhysXFoundation_static_64)
```

Can someone help please ?

AlesBorovicka, 2021-06-13:
Hi,
 you need to add pthread and dl to you target link libraries. 
See for example the snippets cmake file: [https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/snippets/compiler/cmake/linux/SnippetTemplate.cmake#L80](url)

Regards,
Ales

Bicyclette, 2021-06-23:
Hi,

sorry for the late reply !
Thank you for the help, now it works fine :)

Regards,
Mathias

closed 2021-06-23 by Bicyclette