NVIDIAGameWorks/PhysX #433: Alternatives to Trigger on Trigger contact reporting

TheDonsky, 2021-06-13:
PxSimulationEventCallback::onTrigger's documentation states: "Trigger shapes will no longer send notification events for interactions with other trigger shapes".

Although rare, I can easily come up with a number of use cases, when I actually may need to know when two trigger shapes are touching. I can technically "hack" the thing by creating a "special" type of a collider that simply does not interact with any other non-trigger shape, but still reports trigger touches, but that does not seem like a clean enough solution to me and would prefer if the game logic did not concern itself with such issues.

Are there any alternative ways to capture all trigger-on-trigger overlaps that include active bodies?

PierreTerdiman, 2021-06-14:
There is a snippet demonstrating how to emulate trigger shapes, to get both CCD for triggers and trigger-trigger interactions:

https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/snippets/snippettriggers/SnippetTriggers.cpp#L32

This would be the official way to achieve this.

TheDonsky, 2021-06-14:
Thanks! I'll look into the implementation;
One more thing: my use case is for the physics toolbox for my game engine  and I'm hiding most of the abstraction behind some simplified interfaces and the game can not be particularly selective about simulating the triggers or using the built in ones; So, naturally, I have some performance considerations. If I were to simulate all the shapes tied to dynamic bodies and let the static colliders be built-in triggers, I can rest assured that the performance impact will be negligible, right?

PierreTerdiman, 2021-06-14:
I"m afraid I cannot answer a performance question like this - I know nothing about your use case, how many static colliders you have, how many objects will interact with them, etc, etc, etc, it's impossible to say. All I can report is that people using a lot of triggers have seen a measurable performance boost when dropping built-in triggers and switching to emulated triggers. Whether it's negligible or not depends on how much your physics budget is for the frame, and what is happening exactly is that physics time - I cannot answer that. Gut feeling is that while it could be measurable, it is unlikely to significantly change your framerate.

closed 2021-06-14 by TheDonsky