NVIDIAGameWorks/PhysX #436: PxJoint::getRelativeAngularVelocity error

asdasdasdasdasdf, 2021-06-25:
The documentation for this method reads
```
get the relative angular velocity of the joint
This function returns the angular velocity of actor1 relative to actor0. The value is returned in the constraint frame of actor0
```

The code looks like this (I think): https://github.com/NVIDIAGameWorks/PhysX/blob/c3d5537bdebd6f5cd82fcaf87474b838fe6fd5fa/physx/source/physxextensions/src/ExtJoint.h#L235

I believe there are two problems with this method.
1. The documentation states that the result is in the constraint frame of actor0 but the code appears to be in the center of mass pose of actor0.
2. I think it should be calling rotateInv rather than transformInv. It doesn't make sense to apply a positional offset to a velocity.

kstorey-nvidia, 2021-06-28:
I agree with your assessment. Translating a velocity doesn't make sense. Will file a defect.

pkusilence, 2021-06-30:
PxJoint::getRelativeLinearVelocity()	 also returns a invalid results. Beside, the original version calculates the linvel & angvel in the COM frame of actor 0 , instead of the constraint frame 0 (as is claimed in PxJoint.h)。

the fix can be like this: 
`PxTransform	 getRelativeTransform()			const
		{
			PxRigidActor* actor0, * actor1;
			mPxConstraint->getActors(actor0, actor1);
			PxTransform c2b0 = getBodyPose(actor0) * getCom(actor0).transformInv(mLocalPose[0]),
						c2b1 = getBodyPose(actor1) * getCom(actor1).transformInv(mLocalPose[1]);
			return c2b0.transformInv(c2b1);
		}

		PxVec3	getRelativeLinearVelocity()			const
		{
			PxRigidActor* actor0, * actor1;
			PxVec3 l0, a0, l1, a1;
			mPxConstraint->getActors(actor0, actor1);

			//COM to world 
			PxTransform t0 = getBodyPose(actor0),
						t1 = getBodyPose(actor1);
			
			//joint frame to COM
			PxTransform	c2b0 = getCom(actor0).transformInv(mLocalPose[0]),
						c2b1 = getCom(actor1).transformInv(mLocalPose[1]);

			getActorVelocity(actor0, l0, a0);
			getActorVelocity(actor1, l1, a1);

			PxVec3 cv0ToWorld = l0 + a0.cross(t0.rotate(c2b0.p)), 
				   cv1ToWorld = l1 + a1.cross(t1.rotate(c2b1.p));

			PxQuat q0 = t0.q * c2b0.q;

			return q0.rotateInv(cv1ToWorld - cv0ToWorld);
		}

		PxVec3	getRelativeAngularVelocity()		const
		{
			PxRigidActor* actor0, * actor1;
			PxVec3 l0, a0, l1, a1;
			mPxConstraint->getActors(actor0, actor1);

			PxQuat cq0 = getBodyPose(actor0).q * getCom(actor0).q.getConjugate() * mLocalPose[0].q;
			getActorVelocity(actor0, l0, a0);
			getActorVelocity(actor1, l1, a1);

			return cq0.rotateInv(a1 - a0);
		}`

asdasdasdasdasdf, 2021-06-30:
that still erroneously involves the center of mass right? the documentation says it should be in the constraint frame of actor0

pkusilence, 2021-06-30:
> that still erroneously involves the center of mass right? the documentation says it should be in the constraint frame of actor0

Actually , I have already tested the fix and these are right. I’m using the function getCom to get the inverse of com transform, and then it is canceled by com transform included in getBodyPose. So, eventually it doesn’t matter what the com transform is (only for calculating the relative transform)

While calculating relative Lin/Ang velocity, we know that function getactorvelocity returns the com linvel and angvel in world space, rather than actor’s velocities. So in this situation we rely on the com to actor transform, and constraint frame to actor transform, to finally get constraint frame to com transform. Then we can get velocities of the frame.

For PhysX 4 the function getBodyPose is removed,

pkusilence, 2021-07-01:
An other way to fix:

`               PxTransform getGlobalPose(const PxRigidActor* actor) const
		{
			if (!actor)
				return PxTransform(PxIdentity);
			else
				return actor->getGlobalPose();
		}

		PxTransform getComLocal(PxRigidActor* actor) const
		{
			if (!actor)
				return PxTransform(PxIdentity);
			else if (actor->getType() == PxActorType::eRIGID_DYNAMIC || actor->getType() == PxActorType::eARTICULATION_LINK)
				return static_cast<PxRigidBody*>(actor)->getCMassLocalPose();
			else
				return PxTransform(PxIdentity);
		}

                PxTransform	 getRelativeTransform()			const
		{
			PxRigidActor* actor0, * actor1;
			mPxConstraint->getActors(actor0, actor1);
			PxTransform c2b0 = getGlobalPose(actor0).transform(mLocalPose[0]),
						c2b1 = getGlobalPose(actor1).transform(mLocalPose[1]);
			return c2b0.transformInv(c2b1);
		}

		PxVec3	getRelativeLinearVelocity()			const
		{
			PxRigidActor* actor0, * actor1;
			PxVec3 l0, a0, l1, a1;
			mPxConstraint->getActors(actor0, actor1);

			//COM to world 
			PxTransform t0 = getBodyPose(actor0),
						t1 = getBodyPose(actor1);
			
			//joint frame to COM
			PxTransform	c2b0 = getComLocal(actor0).transformInv(mLocalPose[0]),
						c2b1 = getComLocal(actor1).transformInv(mLocalPose[1]);

			getActorVelocity(actor0, l0, a0);
			getActorVelocity(actor1, l1, a1);

			PxVec3 cv0ToWorld = l0 + a0.cross(t0.rotate(c2b0.p)), 
				   cv1ToWorld = l1 + a1.cross(t1.rotate(c2b1.p));

			PxQuat q0 = t0.q * c2b0.q;

			return q0.rotateInv(cv1ToWorld - cv0ToWorld);
		}

		PxVec3	getRelativeAngularVelocity()		const
		{
			PxRigidActor* actor0, * actor1;
			PxVec3 l0, a0, l1, a1;
			mPxConstraint->getActors(actor0, actor1);

			PxQuat cq0 = getGlobalPose(actor0).q * mLocalPose[0].q;
			getActorVelocity(actor0, l0, a0);
			getActorVelocity(actor1, l1, a1);

			return cq0.rotateInv(a1 - a0);
		}
`

asdasdasdasdasdf, 2021-07-01:
yeah that implementation of getRelativeAngularVelocity looks a lot more like what I expected.

msauter-nvidia, 2023-12-11:
A fix is planned to go into the next release (probably 5.3.2-105.2)

shaneasd, 2023-12-12:
Thanks for the update!

preist-nvidia, 2024-03-21:
Closing as will be in next release (now 5.4.0-106.0)

closed 2024-03-21 by preist-nvidia