NVIDIAGameWorks/PhysX #437: PVD renders are broken

Paril, 2021-06-25:
For some reason, PVD's renderer is entirely broken. Not sure if this is a Windows 10 thing or what, but all of the rendered geometry is flat and gigantic. Here's a sample with a bunch of boxes on a plane: https://streamable.com/3yhxz9

[transport.zip](https://github.com/NVIDIAGameWorks/PhysX/files/6717954/transport.zip)

PierreTerdiman, 2021-07-01:
Not sure what's going on. This files renders fine on my Win10 PC.

[image]

Paril, 2021-07-01:
Specifically, this is with an AMD video card (AMD Vega 56). It's probably some kind of incompatibility. I'm not sure offhand what it could be, though.