NVIDIAGameWorks/PhysX #438: Using local pose position offset on shapes for PxArticulationLink breaks everything

jankrassnigg, 2021-06-25:
I am struggling to get a simple shape setup working with articulations.

I can create a number of actors (PxArticulationLink) and give each one a shape (using PxRigidActorExt::createExclusiveShape). I don't set up any joints yet and disable gravity, so after creation the shapes just float in their initial position.

Now I want to adjust the shape transforms Using PxShape::setLocalPose. For rotations that works as expected. However, if I touch the translation, the shapes just rotate even more !
It looks as if there is a memory corruption. Or maybe the translation is somehow added to the rotation.

Is this some kind of limitation for shapes on articulations? Do they not support shape position offsets? I don't see something like that mentioned in the docs. I wanted to use local poses for shapes, so that I can attach multiple shapes to the same link. If that's not possible I guess I can use one link per shape, and move the shape offset there, but then I can't have composite shapes for a single link.

Has anyone else tried this before and succeeded?

jankrassnigg, 2021-06-25:
After a lot of trial and error I figured it out:

I had not configured the joints yet, since I assumed there would be no joints as long as I don't set them up. However, there are always joints, and they are always anchored to the center of BOTH articulation links. That means that the links will immediately be pulled together during the first update step.

I also didn't have any hierarchy set up yet, all links were just created under one root link in a flat list and I had gravity disabled. As a result, all links were pulled together during the first update step, and the overall position of all objects got displaced. However, for some reason, the rotations stayed exactly in their initial state. However, once I introduced a position offset to any of the shapes, this initial pull gave all the shapes a spin, creating a big chaotic heap. I don't understand why that didn't introduce any momentum, though, I would have expected the bodies to keep moving for a while. That didn't happen, which made it even harder to understand.

Also, the PhysX documentation is at times not very precise. It took me quite a while to figure out that the positions given to createLink have to be ALL in global space, never relative to a parent, whereas the joint parent/child poses have to be in local space (that one is mentioned).

closed 2021-06-25 by jankrassnigg

kstorey-nvidia, 2021-06-28:
If you want to create compound shape actors without creating joints, you can attach multiple shapes to the same actor using different shape local poses. This should be more efficient than setting up many articulation links attached with fixed joints.