NVIDIAGameWorks/PhysX #439: How is elapsed simulation time calculated?

morphogencc, 2021-06-27:
I've noticed that when I move an application between computers, the rate at which the elapsed simulation time increases seems to change.  I recently got a new computer with a more powerful CPU and GPU, and I've noticed that the simulation time proceeds about 10x faster than previous.

It definitely is related to how often I call `physx::PxScene::simulate(timeStep)`, but the actual timestep doesn't seem to matter -- if I call it with smaller or larger timesteps (even a factor of 1000), the simulation time proceeds at the same rate -- currently about 20 "world seconds" per real-world seconds.

I've tried manipulating my apps' frame rate to calibrate the time -- I would expect frame rate of 60 fps, and a time step of 1.0/60.0 to create a 1:1 mapping of elapsed simulation time to real-world time, but this isn't the case.

Can anyone elaborate how this elapsed simulation time is calculated?

kstorey-nvidia, 2021-06-28:
Is this with your own application? PhysX doesn't manipulate the time-step that you pass to simulate(timeStep) at all. If time is elapsing faster than real-time, it is because the application is stepping PhysX faster than real-time.

If you look at the PhysX snippets, these simply step PhysX at a fixed time-step and allow v-sync to limit frame rate. If you turn off v-sync, you may get faster-than-realtime simulation.

In the PhysX samples, there is a more advanced time-stepping system that is closer to how a game engine might work. The implementation in the samples should not step faster than real-time, although it can step slower than real-time if the frame drops too low.

morphogencc, 2021-06-28:
Interesting; I'm definitely seeing this behavior in my own application.

How does the `PxScene::getTimestamp()` method compute the timestamp?  Does it simply add the incremental time-steps, or does it use another method to create the timestamp?  Is there a better way to calculate the elapsed world-time in the simulation?

PierreTerdiman, 2021-06-28:
A timestamp is not a timestep. The two are unrelated. The timestamp is just an integer counter incremented with each update. This is mainly used to invalidate potential caches on the user's side. You most likely do not need it.

	/**
	\brief Retrieves the scene's internal timestamp, increased each time a simulation step is completed.

	\return scene timestamp
	*/
	virtual	PxU32				getTimestamp()	const	= 0;

PierreTerdiman, 2021-06-28:
_**Is there a better way to calculate the elapsed world-time in the simulation?**_

Are you looking for something like timeGetTime() ?

morphogencc, 2021-06-30:
@PierreTerdiman Ah, got it -- that makes sense!  It sounds like it's not really what I'm looking for.

I'm looking for a way to easily track the "world" time, not necessarily my system time.  I'd like to be able to fast-forward my simulations, at, say, 10x -- and be able to keep track of the time that's progressing within the world even at 10x speed.  Ideally the Scene would track every dT that I used during the update phase and I could track it; this way if I change my deltaT or just iterate the `simulate()` method, then I could consistently track the world time.

PierreTerdiman, 2021-07-01:
Maybe I'm missing something. Why don't you accumulate the total time yourself, each time you call PxScene::simulate() ?

morphogencc, 2021-07-01:
I can (and that's what I'm currently doing) I just didn't know if the Scene tracked this information itself.