NVIDIAGameWorks/PhysX #441: Creating knee/elbow joints with PxArticulation

jankrassnigg, 2021-06-28:
I'm trying to set up a ragdoll for a human and I'm using the PxArticulation feature for this. Now I need to set proper joint swing and twist limits, however I just noticed that the twist limit on PxArticulationJoint only has a value for the Z and Y limit, but it doesn't have an upper and lower limit for each (as the twist limit has).

That means it would be very hard to build a knee or elbow joint, since they would have say an upper swing limit of zero and a lower swing limit of 120 degree or so. The only solution I see is to place the reference frame for the joint in the middle of these values, which means more painful fiddling with bone transforms.

Is that the way to go, or am I missing something?

kstorey-nvidia, 2022-03-18:
You will need to rotate the joint frame to either place the frame in the middle of your range, or rotate so that it aligns with the twist axis, where lower and upper limits are supported.

There is a method in PxMathUtils.h called PxShortestRotation, that will help you work out the rotation required to line up your twist axis (X) with the axis you want to have a hinge constraint to work around.

MoMoPF, 2022-11-10:
> 

Hi! @kstorey-nvidia 
I also have the similar problem about joint limit cone. Take a spherical joint as an example. If I use PxJointLimitCone but don't make the place in the middle of limit range or rotate so that it aligns with the twist axis, the swinging seems a bit strange that the actor may collide at some place not as expected. I can not figure out what the exact boundary of the rotation is in this situation. Is the frame alignment necessary for PxJointLimitCone ?