NVIDIAGameWorks/PhysX #443: PxMassProperties case eCONVEXMESH error

bingbingzhenbang, 2021-07-01:
When PxMeshScale s has a non-uniform scale (s.scale.x != s.scale.y || s.scale.x != s.scale.z || s.scale.y != s.scale.z) and non-identity rotation, scaledInertiaTensorNonCOM should be calculated first, and inertiaTensor returned should be about the scaled and rotated centerOfMass.
I suggest the corret way should be like this:
```
				const PxConvexMeshGeometry& c = static_cast<const PxConvexMeshGeometry&>(geometry);
				PxVec3 unscaledCoM;
				PxMat33 unscaledInertiaTensorNonCOM; // inertia tensor of convex mesh in mesh local space
				PxMat33 unscaledInertiaTensorCOM;
				PxReal unscaledMass;
				c.convexMesh->getMassInformation(unscaledMass, unscaledInertiaTensorNonCOM, unscaledCoM);
				
				const PxMeshScale& s = c.scale;
				mass = unscaledMass * s.scale.x * s.scale.y * s.scale.z;
				centerOfMass = s.rotation.rotate(s.scale.multiply(s.rotation.rotateInv(unscaledCoM)));

				PxMat33 scaledInertiaTensorNonCOM = scaleInertia(unscaledInertiaTensorNonCOM, s.rotation, s.scale);
				inertiaTensor[0][0] = scaledInertiaTensorNonCOM[0][0] - mass *PxReal((centerOfMass.y* centerOfMass.y+ centerOfMass.z* centerOfMass.z));
				inertiaTensor[1][1] = scaledInertiaTensorNonCOM[1][1] - mass *PxReal((centerOfMass.z* centerOfMass.z+ centerOfMass.x* centerOfMass.x));
				inertiaTensor[2][2] = scaledInertiaTensorNonCOM[2][2] - mass *PxReal((centerOfMass.x* centerOfMass.x+ centerOfMass.y* centerOfMass.y));
				inertiaTensor[0][1] = inertiaTensor[1][0] = (scaledInertiaTensorNonCOM[0][1] + mass *PxReal(centerOfMass.x* centerOfMass.y));
				inertiaTensor[1][2] = inertiaTensor[2][1] = (scaledInertiaTensorNonCOM[1][2] + mass *PxReal(centerOfMass.y* centerOfMass.z));
				inertiaTensor[0][2] = inertiaTensor[2][0] = (scaledInertiaTensorNonCOM[0][2] + mass *PxReal(centerOfMass.z* centerOfMass.x));
```

preist-nvidia, 2021-08-05:
Hi Rubing,

The two approaches are equivalent I think. The code as it is (except for the wrong center of mass computation, as you found [in the other issue](https://github.com/NVIDIAGameWorks/PhysX/issues/457)) produces the correct moment of inertia. Basically, it should not matter whether we first scale and then shift, or first shift and then scale.

Best,
Philipp

bingbingzhenbang, 2021-08-06:
> Hi Rubing,
>
> The two approaches are equivalent I think. The code as it is (except for the wrong center of mass computation, as you found [in the other [...]

I check my calculation, you are right, thank you.

preist-nvidia, 2021-08-06:
Great, thank you for checking.

closed 2021-08-06 by preist-nvidia