NVIDIAGameWorks/PhysX #446: Consider making an Nvidia Engine for Unreal

delebash, 2021-07-04:
Currently to use some of Nvidia features such as flow and flex you have to build from source.  Also no support has been added for any recent versions of Unreal.  Maybe consider making an Nvidia Engine for Unreal like Houdini-Engine https://www.sidefx.com/products/houdini-engine/ this way future developments by Nvidia can be used as assets in Unreal.

alanjfs, 2021-07-12:
Unlike Houdini, PhysX is open source. You could make such a tool yourself and even make a living out of it!

delebash, 2021-07-12:
Yeah don't have that kind of skill!  I just think it would be advantageous for NVidia to be able to use their tools in Unreal.  Clearly at one point they thought so too as they had specific builds of Unreal with flow, blast, etc.  I guess they changed their minds as it seems for any recent version of Unreal that route has been abandoned.