NVIDIAGameWorks/PhysX #447: [Physics.PhysX] Gu::ConvexMesh::checkHullPolygons: Hull seems to have opened volume

Xcrypt1991, 2021-07-06:
I am trying to debug a 'procedurally generated' mesh collider. I get the error [Physics.PhysX] Gu::ConvexMesh::checkHullPolygons: Hull seems to have opened volume or do (some) faces have reversed winding?
I have no idea what this could mean. It seems like each triangle is well connected. I have also created a script that inverts through all possible combinations of triangle windings and I get the error on each iteration.
This happens when I'm trying to make the mesh convex by setting MeshCollider.convex to true (unity)
any ideas?

AlesBorovicka, 2021-07-07:
Hi,
 could you please provide the input vertices that go to the PhysX convex cooking? I tried to use the the vertices that you captured on the screen, but they seems to be fine when provided to our cooking. 
Thanks,
Ales

Xcrypt1991, 2021-07-07:
> Hi,
> could you please provide the input vertices that go to the PhysX convex cooking? I tried to use the the vertices that you captured on the screen, but they seems to be fine when provided to our cooking.
> Thanks,
> Ales

Hello,
I am not sure how PhysX is implemented in Unity, but the script below gives the afforementioned error.
https://pastie.io/soipoy.go

AlesBorovicka, 2021-07-08:
Looks like the tolerance in this test that fails is too low. I did changed the tolerance so that this convex passes the check. I am not sure how to deliver the fix to you. Do you have the source code available?

Xcrypt1991, 2021-07-08:
> Looks like the tolerance in this test that fails is too low. I did changed the tolerance so that this convex passes the check. I am not sure how to deliver the fix to you. Do you have the source code available?

I do not have a license to modify the unity source code. If there is no error on your side, I think this is an issue that unity should fix by for example providing a way to change this tolerance you are talking about.

AlesBorovicka, 2021-07-08:
The issue is on our side, I fixed that on our side, but I am not sure when this gets to Unity. Thats why I asked if you have source code, so that you could change it and unblock yourself.

Xcrypt1991, 2021-07-08:
> The issue is on our side, I fixed that on our side, but I am not sure when this gets to Unity. Thats why I asked if you have source code, so that you could change it and unblock yourself.

Oh I see. Unfortunately I don't have the source code. Don't worry, I will find a workaround while I wait for this to get to unity. Thanks for the support :)