NVIDIAGameWorks/PhysX #448: Need some help building SDK on arch-linux (Fedora 23)

TheDonsky, 2021-07-09:
Hello, we are having trouble building the repository on arch-linux; We've successfully built and used on Debian (Ubuntu) and Windows.
Here are the usual steps we've taken so far:
0. pull the repository;
1. cd physx;
2. ./generate_projects.sh (with 'linux' option);
3. cd compiler/linux-checked;
4. make

Here are a few errors I'm getting from make:
```
/home/Donsky/Documents/PhysX/physx/include/task/PxTask.h:114:10: error: '~PxTask' overrides a destructor but is not marked 'override' [-Werror,-Wsuggest-destructor-override]

        virtual ~PxTask() {}

                ^

/home/Donsky/Documents/PhysX/physx/include/task/PxTask.h:48:10: note: overridden virtual function is here

        virtual ~PxBaseTask() {}

                ^

/home/Donsky/Documents/PhysX/physx/include/task/PxTask.h:117:18: error: 'release' overrides a member function but is not marked 'override' [-Werror,-Wsuggest-override]

    virtual void release()

                 ^

/home/Donsky/Documents/PhysX/physx/include/task/PxTask.h:80:19: note: overridden virtual function is here

    virtual void                release() = 0;

```
(there are more like these, but all of them will likely go away if we find cflags and add something like -Wno-inconsistent-missing-destructor-override; Problem is, we don't know how where to make the changes)

Also, a few weeks back we managed to alter the cmake file so that there errors were gone, but forgot how and even after that we were getting some errors regarding CMAKE_LIBRARY_ARCHITECTURE missing...

Can you help us out?

ayoub-belarbi, 2021-07-09:
Hi @TheDonsky, 
Compiler flags for Linux can be added in `physx\source\compiler\cmake\linux\CMakeLists.txt`.

Please note that CMAKE_LIBRARY_ARCHITECTURE will only work for Debian probably, please check: https://github.com/NVIDIAGameWorks/PhysX/pull/215 and https://github.com/NVIDIAGameWorks/PhysX/issues/39

and check also: https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/externals/cmakemodules/GetCompilerAndPlatform.cmake

Thanks

TheDonsky, 2021-07-09:
Hello @ayoub-belarbi ,
Thanks a lot!

Adding "-Wno-suggest-override -Wno-suggest-destructor-override -Wno-dtor-name" to CLANG_WARNINGS at line 31 helped and we did not get any additional errors this time.

We can make make a pull request if you think it'll be useful

closed 2021-07-09 by TheDonsky