NVIDIAGameWorks/PhysX #452: drives failing to achieve maximum force

asdasdasdasdasdf, 2021-07-16:
I'm simulating a wheel loader. This link shows the basic layout of a wheel loader for illustration purposes though it's not this one that I'm simulating https://wastemanagementreview.com.au/wp-content/uploads/2018/11/wheel-loader-pic.jpg .

Effectively there's a linear drive which pushes on the lever in the middle which in turn pulls on a link connected to the bucket which then rotates. Note that this is implemented using constrained ridid bodies rather than articulations with the PGS solver in physx 4.1.2. The drive itself is a force limited damper with target velocity dictated by user input. There are a few complications over that basic model which I'm glossing over which are hopefully unimportant.

I have a  requirement that when the bucket is on the ground and the hydraulic is contracted such that the bucket is rotated downwards the machine should lift off the ground. i.e. the hydraulic which rotates the bucket should be strong enough to lift the machine by rotating the bucket.

What I am observing is that the machine is failing to lift and yet the force exerted by the drive is a fraction of the maximum force it is permitted to exert. I'd like to know:
- why is this?
- what are my options to fix it?

Changing the damping coefficient has a minimal impact. Changing the target velocity makes a big difference. If I say I want to extend at 1m/s with 10^10N of force then I don't get enough force to move. If I say I want to extend at 2m/s then it moves just fine. These are of course just illustrative values.

My suspicion is that it's to do with the relevant mass ratios in the system. In general the ratios of the various physical parts in the system maintain the 1:10 recommendation but they're not ideal. It's a little harder to tell for moments of inertia but I believe in my case the force pushing on the bucket is only about 9% of the force at the hydraulic (depending on the exact configuration of the system) due to the levers involved. The mass any individual part is certainly less than 10% of the mass of the whole machine which I'm trying to lift.

I believe the TGS solver is intended to do a better job of this and indeed it has solved similar problems in the past for me but it has also caused unacceptable instability (read: explosions) so I conclude that it is inappropriate for my problem.

I'm curious about whether articulations would do a better job of this. Unfortunately it would be quite difficult to restructure everything to use an articulation so I wouldn't want to attempt it unless there was a theoretical expectation it would solve my problem. I'm also concerned about the circular dependencies in the parts in the system and how that would work for an articulation.

From reading the code and documentation, it appears that using the mass/inertia scales should solve my problem. However I've found that if I scale mass/inertia symmetrically across the two bodies then I achieve no effect. This surprises me as it seems like for damping to make so little difference, dt * damping * unitResponse must be must greater than 1 so reducing unitResponse should solve my problem but in (https://github.com/NVIDIAGameWorks/PhysX/blob/c3d5537bdebd6f5cd82fcaf87474b838fe6fd5fa/physx/source/lowleveldynamics/src/DyConstraintSetup.cpp#L478) it appears that unitResponse is directly proportional to the mass/inertia scales.

Applying mass/inertia scales asymmetrically results in a net force on the system as a whole resulting in my machine accelerating faster and faster along the ground without any input.

I could implement some sort of feedback loop where I artifically apply additional force or increase target speed when I observe that neither the target speed nor maximum force are reached. However I'd prefer to avoid this because of the weird side effects it might have in cases where things would otherwise work fine.

preist-nvidia, 2021-07-16:
Hi asdasdasdasdasdf,

Why are you using a velocity drive and not a position drive? I think a position drive would be more natural in this application. Of course it might move too quickly for big target position jumps, but you could use appropriate damping or use a time-sampled trajectory for the target positions.

The velocity drive force will be proportional to the velocity error, and so if the machine is at rest, it will be limited by the magnitude of the target velocity and the damping parameter (they should have the same impact on your dynamics however).

Are you computing the forces yourself? To get a position drive, you could use the equation at the beginning here: https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/Joints.html#drives

Alternatively, you could use D6 joints that implement the drive for you. You can then consider using the acceleration drive option that should make tuning easier.

Using articulations should work, too, if the maximal coordinate approach does not work out.

I hope this helps,
Philipp

asdasdasdasdasdf, 2021-07-16:
Hi Philipp,
Thanks for your response.

I should clarify that the user input is a target velocity rather than a target position. I want the applied force to be proportional to the velocity error rather than the positional error because under external load I want to hit an equilibrium velocity rather than equilibrium position. i.e. if the hydraulic has to work hard I want it to travel at half the speed rather than stop at half its extension. I could always target full extension or full contraction depending on the user input and then set the spring coefficient to be very large to render the actual position error inconsequential. As you say, I'd have to then provide damping to stop it going too fast.

for simplicity I use vT and xT to indicate target values.

Currently I have F = -b(v - vT)         clamped to Fmax
I could instead have F = -k(x - xT) - b(v-vT)        clamped to Fmax
but k would have to be very large to avoid the force being proportional to position error at which point b would have to be very large to have any effect which would prevent me getting the relationship between force and velocity that I want.

With regard to your second point, I agree that the velocity drive should be limited by target velocity, actual velocity, damping and max force. In the situation I describe, actual velocity is zero (effectively) target velocity is moderate, damping is large and max force is large. Specifically ```damping * (target velocity - velocity) > max force``` so I would expect the  actual force to be the max force. Also since I'm not hitting the max and velocity is zero, I would expect the force to be F = damping * target and as such I agree they should have the same impact on dynamics. This is not the case. From the code, I believe this is expected in cases where ```unitResponse != 0``` but I'm not clear on the basis for that deviation from theory (i.e. I don't know what that value is supposed to represent).

I'm not sure what you mean about computing the forces myself. I have configuration which specifies what the damping coefficient, target speed and max force should be. I don't calculate the actual force myself (other than to investigate this bug). In general my experience has been that when the dynamics I want can be correctly modelled with a spring-damper it is better to allow the physics engine to do it for me.

I'm not clear on whether/how acceleration drives would help me. My understanding is that instead of ```F=-b(v-vT)``` they modeled ```a=-b(v-vT)``` and then multiplied the result by mass to get the force. I already have the relevant values so an acceleration drive would just require me to divide everything by mass and end up with the same result. Perhaps I'm missing something?

Could you elaborate on how articulations would improve the situation. Are they theoretically superior in a way that is pertinent to my problem? How would I resolve the issue of circular dependencies?

Fundamentally my issue is that, contrary to my expectation, the force generated by physx from a drive that claims to be defined as ```F = -b(v - vT) clamped to FMax```, is substantially less than ```-b(v - vT)``` and also substantially less than FMax. Do you have any insight as to why this would be?

preist-nvidia, 2021-07-16:
Which PhysX joint and drive are you using? A D6?

Could you verify the mismatch between expected and realized force on a simple example? E.g. moving a block up against gravity which should only start moving up at the point where Fmax and damping * targetVel are larger than m * g?

asdasdasdasdasdf, 2021-07-16:
I'm using a hybrid between a D6 and distance joint. Effectively the solver prep is done like a D6 but the axis is selected like a distance joint.

I'll come up with a simpler test case using a D6. I suspect it will require multiple bodies but I'll see.

preist-nvidia, 2021-07-16:
Sounds good - I hope the simplified example can shed some light on the issue.

asdasdasdasdasdf, 2021-07-16:
I wasn't able to reproduce the issue in a simple case (a cube attached to the world with a D6Joint velocity drive and also to a much heavier cube).

However I have dug deeper into my current case. It turns out that the reported velocity _is_ close enough to the target that there is insufficient force to lift the machine. However the reported velocity is not accurate when compared to the actual change in position. i.e. if I use getVelocityAtPos to get the velocity at the two end points of the distance drive and compare to the actual rate of change of the position of the two end points I find that the relative velocity of the joint is around 0.017m/s (comparable to the target) whereas the actual change in relative position of the points is only 0.000002m/s (basically noise).

How can the velocity be distinctly non-zero without the position actually changing? One thing that springs to mind is joint projection. I'm not doing anything special here with projection but I believe the standard D6 joints I'm using do do some amount of projection. Could that be the cause? Could it be something else?

preist-nvidia, 2021-07-19:
The TGS solver produces noisier velocities than the PGS solver (but I am not sure this would explain why the constraint is not working well) - so you could try to see what happens if you use PGS.

Otherwise maybe it is worth exploring if an articulation produces more accurate results in your case - we do recommend using articulations for robotic mechanisms. While the articulation cannot contain loops directy, you can close loops using regular rigid-body joints (that you have been using). Articulation prismatic joints also support adding both position and velocity drives.

See here for differences between max. coordinate vs reduced coordinate articulations: https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/Articulations.html#maximal-coordinate-and-reduced-articulations

asdasdasdasdasdf, 2021-07-19:
I'm currently using PGS. I found TGS caused severe stability problems in other circumstances.

I'll have a read of that article. Thanks.

Can you offer any insight as to why the actual displacement is not in line with the reported velocity for these bodies in PGS? Note that at this point it's independent of the joints used. I can see from querying the bodies themselves that the displacement over time is zero but the "velocity" is non-zero.

preist-nvidia, 2021-07-19:
I am not sure about the origins of the discrepancy. Another thing to try would be to make sure that the solver converges - have you tried raising the position iterations of the PGS solver?

asdasdasdasdasdf, 2021-07-19:
I'm currently running with 100 position and 100 velocity iterations. Is there any output from the engine which gives an indication of whether it's converged or not?

kstorey-nvidia, 2021-07-19:
If you can provide a PVD capture of the configuration that you think should be providing more drive force than it is, we might be able to take a look. 

>>I found TGS caused severe stability problems in other circumstances.

If you haven't reported these already, could you report and provide repros for these cases. If you have already, could you link to the problems. There may be things we can do to help here. The TGS solver does converge much faster, especially with strong drive targets. If you can use TGS, it converges much better than PGS.

>I can see from querying the bodies themselves that the displacement over time is zero but the "velocity" is non-zero.

Position iterations are used to produce a velocity used to integrate the pose of the actors. Velocity iterations run after this and compute the velocity that we carry over to next frame. Position solver iterations solve a biased constraint (includes joint/contact error terms), while velocity iterations do not include the joint/contact error terms. This means that the positional change of the rigid bodies is not completely coupled to the velocity that the solver reports to the user. They are usually very close, but under some circumstances where the error terms are the dominant factor in the velocities produced by the position iterations, they could be significantly different.

In general, I wouldn't recommend running a large number of velocity iterations. They do not converge as well as position iterations because error terms are quite often dominant in hard-to-solve cases. In fact, in some cases, they make convergence worse. They are mostly useful to reduce how violent the response is to a penetration. Limiting maxDepenetrationVelocity is usually enough. I'd usually recommend leaving velocity iterations at their default iteration count.

asdasdasdasdasdf, 2021-07-19:
Thanks @kstorey-nvidia 

I'll see what I can do about getting some PVD captures but I may run in to trouble with sharing intellectual property. I understand this would impede your ability to assist me.

I'll have a read through the code on velocity and position iterations and see if I can make sense of the information you've provided but could you please clarify a couple of points? 

What exactly are the joint error terms? Do you refer to the difference between the target and actual velocity/transform? I can engineer situations where related problems occur without contacts so I'm less concerned about them.

What is the nature of the bias when solving a biased constraint? I noticed a coefficient named ```unitResponse``` in the equations which seemed to cause a deviation from a simple implicit solution. Is this the bias term. It looks like it's initialized to the energy of the two jointed bodies but I couldn't figure out why.

I'm not clear on why velocity iteration would be required. It seems to me that all forces not coming from joints/contacts are constant within a physx tick so the position/velocity should be solvable algebraically (and probably already were when solving the joints). Without joints/contacts there's nothing linking the bodies so I don't see why iteration would be required. Clearly it is so I must be missing something.

It sounds like maybe I should try reducing velocity iterations. I'll give that a go.

kstorey-nvidia, 2021-07-19:
If the concern is about sharing securely, we can provide secure FTP credentials for you to send PVD captures to rather than uploading to public GitHub.

What constitutes an error depends on the type of joint. 

For a spring joint, the bias is the combination of velocity target error and position error scaled by their respective damping and spring coefficients. These are carried over and constant between position and velocity iterations.

For a hard constraint, such as a limit or locked axis, the bias is the positional/angular error scaled by effective mass and invDt during position iterations and zero otherwise.

The main reason iteration is needed with systems of rigid bodies and joints is that each joint is solved "locally" using just the mass/inertia of the rigid bodies that are constrained by the joint. A singular joint is solved perfectly by this approach, but any other joints influencing the same bodies solved after this constraint can introduce violations to this constraint.

If you use articulations, this is far less of an issue because native articulation joints are guaranteed to be free from any constraint error.

asdasdasdasdasdf, 2021-07-19:
ah I see so the bias captures the effective inertia at either end of the joint. i.e. it corrects for the fact that to move a body you don't just have to shift its mass but also the mass of everything attached to it. Presumably this is why there is the recommendation to avoid low mass bodies sandwiched between high mass bodies.

I'll keep investigating and get back to you. Thanks.

asdasdasdasdasdf, 2021-07-21:
I'm attempting to set up PVD capture but I'm getting an assertion violation here https://github.com/NVIDIAGameWorks/PhysX/blob/c3d5537bdebd6f5cd82fcaf87474b838fe6fd5fa/physx/source/pvd/src/PxPvdDataStream.cpp#L508
because here https://github.com/NVIDIAGameWorks/PhysX/blob/c3d5537bdebd6f5cd82fcaf87474b838fe6fd5fa/physx/source/pvd/src/PxPvdImpl.h#L104 it can't find the entry
mNamespace: physx3
mName: PxD6Joint

Note that I am using physx 4 but it looks like it still treats the namespace as physx3

kstorey-nvidia, 2021-07-21:
Make sure you called PxInitExtensions and passed the pvd instance to it. This is necessary to register the metadata for the joint types.

asdasdasdasdasdf, 2021-07-21:
Thanks I wasn't doing that and have now added it. Doesn't seem to have fixed the problem though. Are there any other steps I might be missing?

asdasdasdasdasdf, 2021-07-26:
any other advice?

kstorey-nvidia, 2021-07-26:
Are you getting the exact same crash callstack after calling PxInitExtensions? Are you definitely passing an initialized PVD instance and not a null ptr to that method?

asdasdasdasdasdf, 2021-07-26:
Yes I'm calling PxInitExtensions passing in an initialized pvd.

The crash I'm getting is ```PX412_ASSERT(success);``` where cls is ```physx3``` ```PxD6Joint```
```
	virtual PvdError createInstance(const NamespacedName& cls, const void* instance)
	{
		PX412_ASSERT(isInstanceValid(instance) == false);
		PX412_ASSERT(mStreamState == DataStreamState::Open);
		bool success = mMetaDataProvider.createInstance(cls, instance);
		PX412_ASSERT(success);
		(void)success;
		return boolToError(handlePvdEvent(CreateInstance(toStream(cls), toStream(instance))));
	}
```

I believe that's the crash I was getting before. At the very least it's in the same place and with the same arguments

kstorey-nvidia, 2021-07-26:
Can you try running the PhysX samples or snippets with PVD connected to see if you reproduce the same issue? I'm not able to reproduce this issue myself. For example, running SnippetJoint?

asdasdasdasdasdf, 2021-07-26:
I will try this tomorrow and get back to you. Thanks for your ongoing help.

An added complication which I'm exploring now is that I've needed to make extensive modifications to the physx namespaces in order to be able to link against physx 3.3.4 and physx 4.1 without ODR violations. Rechecking my code it seems like JointConnectionHandler should have been renamed but wasn't. I'm not sure whether this is causing problems but it seems suspiciously related. I'll follow that thread as well and get back to you.

asdasdasdasdasdf, 2021-07-27:
Aha progress! My issue was indeed the repeated definitions of JointConnectionHandler between my clones of physx 3 and 4. I believe putting this class in an anonymous namespace would solve this problem. I would gladly submit a pull request if that would help get this resolved for future versions. In the mean time I'll modify my clone. It looks like there might be a handful of similarly scoped classes which I'll have to investigate.

asdasdasdasdasdf, 2021-07-27:
I'm still generating invalid pvd files but the tcp connection seems to work fine so I've used that and exported the resulting pxd2. You mentioned providing ftp credentials for me to upload the file? It's about 60MB.

Peculiarly distance joints aren't showing up in the capture which will likely limit its usefulness because the system in question involves a distance joint directly pushing the bucket and a distance drive (custom joint based off distance joint) which provides the actual impetus for movement. Neither of these are present in the capture. However it does show the zbar having a non-zero linear and angular velocity not reflected in its displacement over time.

These exist in an attempt to minimize the number of low mass bodies in the chain but I might try replacing the drive with a drive on the prismatic (actually a D6) connecting the two ends of the hydraulic so that we can see the forces coming from it.

asdasdasdasdasdf, 2021-07-30:
keen to upload my pvd if you can provide that ftp link

kstorey-nvidia, 2021-07-30:
Ah, sorry. Can you give me an email address to send the FTP account detail to?

asdasdasdasdasdf, 2021-07-30:
sure please email stapp@immersivetechnologies.com

asdasdasdasdasdf, 2021-07-30:
I've uploaded the pxd2 to the FTP as instructed.

asdasdasdasdasdf, 2021-08-04:
any luck?

Sorry to nag. I'm reaching a tipping point where I need to solve this properly or start introducing ugly hacks. Not your problem of course. Just providing context for my nagging ;).

kstorey-nvidia, 2021-08-04:
I'm sorry, but I haven't been able to investigate thoroughly yet. 

My gut reaction is that this is something that would very easily be solved using a Featherstone articulation rather than a set of rigid bodies and joints. 

From I can see that there is a heavy scoop, some lighter bodies which are actuated that cause the scoop to lift. My understanding is that we have the bucketTiltRod being pushed forward, causing the scoop to lift.

The issue is likely to relate to the implicit spring model in the drives that attempts to avoid very stiff drives from becoming unstable. With this approach, you can set incredibly stiff springs that will remain stable, where an explicit spring would likely become unstable. The drives are computed based on the mass of the bodies attached by the joints.

When the drives operate on simple rigid bodies, the masses used are the exact mass of the respective bodies. If there are a bunch of other rigid bodies attached to one of the bodies, meaning the drive force required will be much larger, the drives will not work brilliantly. They should eventually reach the required force over many iterations, assuming stiffness/damping and force limits allow that force to be reached, but it can take a lot of iterations. 

The rods have masses of 340Kg, while the Boom has a mass of 4990 and bucket has a mass of 6850. There are inertias and offsets involved, so the effective mass ratios may be even larger. The problem is likely that the drive forces are being computed based on these 340Kg masses and the responses are being driven towards becoming a hard constraint and not reaching the strengths that you expect.

When the bodies involved are articulations, the masses used in the formula are the full effective mass of the multi-body that is attached by the constraint, rather than just that of an individual body. Essentially, the solver knows the exact masses that the drive needs to move and this allows the drive to calculate a much more accurate set of forces so the problem can be solved in a much smaller number of iterations.

You may find that you get closer to where you want to be with rigid bodies if you run at much smaller time-steps. The reason is that the spring formulas incorporate a dt^2 element for positional drive stiffness, so as dt tends towards 0, the forces that can be applied before they are saturated by the implicit drive model become larger. It will still be nowhere near as good as the articulations could be, but it will be better.

The TGS solver is also likely to handle this kind of drive problem a lot better than PGS but you mentioned instabilities. I am fairly certain these instabilities can be solved - in fact, I'd be inclined to suggest they are probably already solved in PhysX 5 but the solver in PhysX 5 is quite different to PhysX 4.

asdasdasdasdasdf, 2021-08-04:
Thanks that's actually really helpful. I'm actually directly driving the zbar from the chassis using a distance drive to avoid the low mass rod/cylinder but there's still a very heavy bucket at the end of the chain and the linkages provide pretty unfavorable leverage at this orientation. This was essentially my assumption about why it wasn't working so it's could to have confirmation.

It's also very useful to have explicit advice that the featherstone articulation will address the root cause of this problem. I think I'll have a much easier time to get approval to investigate this option if it's the recommended solution.

I did consider shorter timesteps but I'm concerned partly about performance and partly about the impact this would have on the rest of the simulation. I'll keep it in mind though. I'm already using 100 position iterations and you advised me that more velocity iterations wouldn't help. I tried 250 position iterations and the improvement was pretty small so I don't think more iterations is the answer in this case.

I'll try and get a PVD capture of the TGS explosion but at this point I'm unlikely to upgrade if I get articulations working. I'd love to be using PhysX 5 but it's not widely available (and the development cost to upgrade is probably prohibitive at this point anyway).

asdasdasdasdasdf, 2021-08-05:
Actually do you have any advice on how to structure my articulation to avoid this problem? My first thought was to essentially replicate my current setup going from high mass to low mass so something like
```
Chassis
 | - Boom
       | - zbar
       | - bucket
```
and then keep the distance drive pushing the zbar away from the chassis and the distance joint maintaining the distance between the zbar and the bucket.

However I'm concerned that the zbar will look like a leaf node and its rotational relationship with the bucket won't be properly understood by the articulation.

Perhaps I'd be better off with something like
```
Chassis
 | - Boom
       | - zbar
           | - Link
              | - Bucket
```
but then the hinge between the bucket and the boom is not captured.

My understanding is that I can't have closed loops and I'm concerned that introducing a regular joint in the wrong place will leave me with the same problem.

kstorey-nvidia, 2021-08-05:
A regular joint with an articulation will be the most likely point of failure, but even with regular joints, the articulation does still offer advantages. The regular joint now uses the total effective mass of the whole articulation rather than just the mass calculated from the individual link.

As a general warning, I favour joints such as distance or d6/spherical without locked angular constraints for these kinds of problems. While the attachment should be with a hinge, the correct degrees of freedom were probably already restricted by the structure of the articulation, and reinforcing, for example, the the joint is revolute can sometimes lead to instabilities because the effective mass around the locked axes for loop constraints like this could be infinite.

asdasdasdasdasdf, 2021-08-05:
Am I correct in saying that with this layout a drive connecting the Chasis and zbar would not get the correct effective mass because it is oblivious to the distance joint connecting the zbar and bucket?
```
Chassis
 | - Boom
       | - zbar
       | - bucket
```

I'm not sure I understand your warning. Is your point that I should avoid redundantly constraining degrees of freedom? i.e. when I close the loop I don't need to limit rotation because that degree of freedom is already removed by the existing hinges?

kstorey-nvidia, 2021-08-05:
I agree that, in this case, the best bet might be to organise the bucket to be a child of the z-bar as this will make the relationship a lot clearer to the drive. In this case, the hinge between the bucket and the boom is not captured in the structure of the articulation and must be implemented with a regular joint, but this should have minimal impact on the simulation.

> I'm not sure I understand your warning. Is your point that I should avoid redundantly constraining degrees of freedom? i.e. when I close the loop I don't need to limit rotation because that degree of freedom is already removed by the existing hinges?

There is redundancy, but it's a bit worse than just a bit of extra work. If you apply a constraint around an axis that is completely locked due to the structure of the articulation, the joint will detect that it needs to apply an infinite force to enforce the constraint. The solver detects this and drops the redundant constraint. However, if the axes don't line up due to errors in the joint frame calculation, this can lead to very large - but not infinite - effective masses being calculated. It's sometimes quite hard to differentiate between a redundant/degenerate constraint and simply a constraint connecting two extremely high mass bodies. However, if you do engineer a degenerate constraint, this can end up applying very large torques around axes that almost line up with locked degrees of freedom. This could potentially lead to simulation instabilities, although it is often OK.

asdasdasdasdasdf, 2021-08-05:
Cool thanks I'll structure my heirarchy like that then.

Thanks I had already learned the hard way that degenerate constraints were bad news but it's good to have it confirmed in this context.

asdasdasdasdasdf, 2021-08-24:
I can confirm that switching to reduced coordinate articulations has solved the problem. Thanks a lot for your help. I still plan to collect a pvd file for the old solution using TGS and will update you when I get it.

asdasdasdasdasdf, 2021-09-09:
my attempts to recreate the exposion with TGS have failed. At some point in the future I imagine I will experiment with TGS again and if I see more problems I'll raise them.

Thanks for your help with this issue.

closed 2021-09-09 by asdasdasdasdasdf