NVIDIAGameWorks/PhysX #456: how to enable PxSimulationEventCallback callback

LAX1DUDE, 2021-08-03:
Hi, I'm trying to get the PxSimulationEventCallback working using it's onContact function so I can detect collisions between specific rigid bodies but it's not being called. ever. I'm running out of debug options. Example source:
```
class PxSimulationEventCallbackImpl : public PxSimulationEventCallback {
public:
	virtual void onConstraintBreak(PxConstraintInfo* constraints, PxU32 count) {

	}

	virtual void onWake(PxActor** actors, PxU32 count) {

	}

	virtual void onSleep(PxActor** actors, PxU32 count) {

	}

	virtual void onContact(const PxContactPairHeader& pairHeader, const PxContactPair* pairs, PxU32 nbPairs) {
		printf("%p %p\n", pairHeader.actors[0], pairHeader.actors[1]);
	}

	virtual void onTrigger(PxTriggerPair * pairs, PxU32 count) {

	}

	virtual void onAdvance(const PxRigidBody* const* bodyBuffer, const PxTransform* poseBuffer, const PxU32 count) {

	}

};
```
And then I pass it to the scene like this:
`sceneDesc.simulationEventCallback = new PxSimulationEventCallbackImpl();`

LAX1DUDE, 2021-08-03:
I used this function from one of the snippets but it only detects collisions between dynamic rigid bodies, I need to detect when the entity is collided with a static rigid body
```
PxFilterFlags contactReportFilterShader(PxFilterObjectAttributes attributes0, PxFilterData filterData0,
	PxFilterObjectAttributes attributes1, PxFilterData filterData1,
	PxPairFlags& pairFlags, const void* constantBlock, PxU32 constantBlockSize)
{
	// all initial and persisting reports for everything, with per-point data
	pairFlags = PxPairFlag::eSOLVE_CONTACT | PxPairFlag::eDETECT_DISCRETE_CONTACT
		| PxPairFlag::eNOTIFY_TOUCH_FOUND
		| PxPairFlag::eNOTIFY_TOUCH_PERSISTS
		| PxPairFlag::eNOTIFY_CONTACT_POINTS;
	return PxFilterFlag::eDEFAULT;
}
```

kstorey-nvidia, 2021-08-04:
Do you mean between pairs of static bodies? That is not something that is supported by PhysX.

You can generate collisions between dynamic and static bodies. Your code should already be generating those.

PhysX has a special classification of dynamic bodies called "kinematic" bodies, set by raising the eKINEMATIC flag on the dynamic body. This imbues the dynamic body with infinite mass. The body can now only be moved by the application setting targets. If there are no targets, the kinematic body behaves identically to a static body.

By default, we will not generate contacts between kinematics and statics because these contacts are not possible to solve (collisions between bodies with infinite mass cannot be solved). However, these can be enabled with:

PxSceneDesc::kinekineFilteringMode
PxSceneDesc::staticKineFilteringMode

Hope this is helpful

LAX1DUDE, 2021-08-05:
I am looking to detect collisions between specific dynamic bodies and specific static bodies. Since my last post I gave up using static bodies and used kinematic bodies (dynamic bodies with the eKINEMATIC flag) instead for my static objects to get the collision events between them and the dynamic bodies in the scene to show up properly in the collision callback. I was originally looking to stick to static bodies but if they need to be kinematic for the collision callback then that's ok because like you said they behave the same on the outside

closed 2021-08-13 by LAX1DUDE